gregdavill/KiBuzzard #1: FilteNotFoundError: ../deps/buzzard/output.scr

pdp7, 2021-01-12:
Thanks for posting this on Twitter.  I ran into this error.  Any ideas?
[image]
```Application: Pcbnew

Version: (5.99.0-8345-gc09dc5abdd), debug build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 GnuTLS/3.6.13 zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.11.0-rc1 x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jan 11 2021 13:09:47
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCE: 6.9.1
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=OFF
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCE=ON
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE=OFF
```
```
pdp7@x1:~/dev/kicad/kicad$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pdp7@x1:~/dev/kicad/kicad$ git log -1
commit c09dc5abdda3d7bb6e21b1574c1695828bf35884 (HEAD -> master, origin/master, origin/HEAD)
Author: Seth Hillbrand <seth@kipro-pcb.com>
Date:   Mon Jan 11 11:23:26 2021 -0800

    Prevent locking items in footprint editor
    
    Adding footprints to the board now has a preference for whether we want
    pads locked or unlocked by default
pdp7@x1:~/dev/kicad/kicad$ 
```

gregdavill, 2021-01-12:
It's probably failing to run the buzzard script. But I'm not detecting this.
When you cloned this repo did you pull down the submodule?

Can you try 
```bash
 git submodules update --init
```

pdp7, 2021-01-13:
@gregdavill I did not do that and have run that command now.  There are more files:
```
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins/KiBuzzard$ find . -type f |fgrep -v .git
./KiBuzzard/__init__.py
./KiBuzzard/__pycache__/__init__.cpython-38.pyc
./KiBuzzard/dialog/__pycache__/dialog.cpython-38.pyc
./KiBuzzard/dialog/__pycache__/dialog_base.cpython-38.pyc
./KiBuzzard/dialog/dialog_base.py
./KiBuzzard/dialog/dialog.py
./__init__.py
./__pycache__/__init__.cpython-38.pyc
./LICENCE
./deps/buzzard/requirements.txt
./deps/buzzard/makeLabel.ulp
./deps/buzzard/buzzard.py
./deps/buzzard/tests/collection.txt
./deps/buzzard/modules/__init__.py
./deps/buzzard/modules/__pycache__/__init__.cpython-37.pyc
./deps/buzzard/modules/__pycache__/svgstring2path.cpython-37.pyc
./deps/buzzard/modules/svgstring2path.py
./deps/buzzard/typeface/__init__.py
./deps/buzzard/typeface/FredokaOne.py
./deps/buzzard/typeface/__pycache__/Roboto.cpython-37.pyc
./deps/buzzard/typeface/__pycache__/__init__.cpython-37.pyc
./deps/buzzard/typeface/__pycache__/FredokaOne.cpython-37.pyc
./deps/buzzard/typeface/Roboto.ttf
./deps/buzzard/typeface/FredokaOne.ttf
./deps/buzzard/typeface/Roboto.py
./deps/buzzard/README.md
./settings_dialog.fbp
./README.md
```
But I still get the same error.
[image]

Do I have it in the right location?  Should I move the inner KiBuzzard up to the parent?
```
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins$ ls -la
total 20
drwxrwxr-x  5 pdp7 pdp7 4096 Jan 12 12:17 .
drwxrwxr-x  3 pdp7 pdp7 4096 Sep 18 06:03 ..
drwxrwxr-x  6 pdp7 pdp7 4096 Jan 12 12:17 KiBuzzard
drwxrwxr-x  5 pdp7 pdp7 4096 Jan  8 13:00 kicad_scripts
drwxrwxr-x 11 pdp7 pdp7 4096 Jan  8 13:00 RF-tools-KiCAD
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins$ ls -la KiBuzzard/
total 56
drwxrwxr-x 6 pdp7 pdp7  4096 Jan 12 12:17 .
drwxrwxr-x 5 pdp7 pdp7  4096 Jan 12 12:17 ..
drwxrwxr-x 3 pdp7 pdp7  4096 Jan 12 12:17 deps
drwxrwxr-x 9 pdp7 pdp7  4096 Jan 12 19:29 .git
-rw-rw-r-- 1 pdp7 pdp7    24 Jan 12 12:17 .gitignore
-rw-rw-r-- 1 pdp7 pdp7   113 Jan 12 12:17 .gitmodules
-rw-rw-r-- 1 pdp7 pdp7    29 Jan 12 12:17 __init__.py
drwxrwxr-x 4 pdp7 pdp7  4096 Jan 12 12:17 KiBuzzard
-rw-rw-r-- 1 pdp7 pdp7  1092 Jan 12 12:17 LICENCE
drwxrwxr-x 2 pdp7 pdp7  4096 Jan 12 12:17 __pycache__
-rw-rw-r-- 1 pdp7 pdp7   282 Jan 12 12:17 README.md
-rw-rw-r-- 1 pdp7 pdp7 11504 Jan 12 12:17 settings_dialog.fbp
```

flummer, 2021-01-13:
Maybe try to run the `buzzard.py` file stand-alone, it might need some dependencies from `./deps/buzzard/requirements.txt`

gregdavill, 2021-01-13:
Yep, I think @flummer might be right. It's likely that buzzard is hitting an error, and we're not handling it correctly.

```console
greg@greg-xps:~/Projects/KiBuzzard$ cd deps/buzzard/
greg@greg-xps:~/Projects/KiBuzzard/deps/buzzard$ python buzzard.py "test"
greg@greg-xps:~/Projects/KiBuzzard/deps/buzzard$
```

If you see an error you'll probably need to install any python deps for buzzard.

```console
greg@greg-xps:~/Projects/KiBuzzard/deps/buzzard$ pip3 install -r requirements.txt --user
```

pdp7, 2021-01-13:
ah, looks like I was missing svgpathtools:
```
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins/KiBuzzard/deps/buzzard$ python3 buzzard.py "test"
Traceback (most recent call last):
  File "buzzard.py", line 1, in <module>
    from svgpathtools import Line, QuadraticBezier, CubicBezier, Path, Arc, svg2paths2
ModuleNotFoundError: No module named 'svgpathtools'
```
```
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins/KiBuzzard/deps/buzzard$ pip3 install -r requirements.txt --user
Collecting bezier==2020.5.19
  Downloading bezier-2020.5.19-cp38-cp38-manylinux2010_x86_64.whl (1.4 MB)
     |████████████████████████████████| 1.4 MB 1.9 MB/s 
Collecting freetype-py==2.1.0.post1
  Downloading freetype_py-2.1.0.post1-py2.py3-none-manylinux1_x86_64.whl (883 kB)
     |████████████████████████████████| 883 kB 3.7 MB/s 
Collecting numpy==1.18.3
  Downloading numpy-1.18.3-cp38-cp38-manylinux1_x86_64.whl (20.6 MB)
     |████████████████████████████████| 20.6 MB 4.1 MB/s 
Collecting svgelements==1.0.2
  Downloading svgelements-1.0.2-py2.py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 3.2 MB/s 
Collecting svgpathtools==1.3.3
  Downloading svgpathtools-1.3.3-py2.py3-none-any.whl (50 kB)
     |████████████████████████████████| 50 kB 3.0 MB/s 
Collecting svgwrite==1.4
  Downloading svgwrite-1.4-py3-none-any.whl (66 kB)
     |████████████████████████████████| 66 kB 1.8 MB/s 
Installing collected packages: numpy, bezier, freetype-py, svgelements, svgwrite, svgpathtools
Successfully installed bezier-2020.5.19 freetype-py-2.1.0.post1 numpy-1.18.3 svgelements-1.0.2 svgpathtools-1.3.3 svgwrite-1.4
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins/KiBuzzard/deps/buzzard$ python3 buzzard.py "test"
```
:)

pdp7, 2021-01-13:
I can run the dialog but not silk label seems to result.  Any ideas?
I tried `"[ SDA)" -a cc`
https://user-images.githubusercontent.com/313864/104511558-c2217400-55a1-11eb-9870-11a1b529e757.mp4

gregdavill, 2021-01-13:
You're likely hitting: https://gitlab.com/kicad/code/kicad/-/issues/7065

Right now the plugin isn't very useful because of this bug.
If you save/close/reopen pcbnew does the label show up?
The fix I've shown in that issue "works" but isn't a production fix and might cause other issues.

A better workflow right now is likely using the clip-board. Try this:
```console
greg@greg-xps:~/Projects/KiBuzzard$ git submodule update --remote
Submodule path 'deps/buzzard': checked out 'b59e46e1e7f356e00668a89008306dbe91dad0de'
greg@greg-xps:~/Projects/KiBuzzard$ cd deps/buzzard/
greg@greg-xps:~/Projects/KiBuzzard/deps/buzzard$ python buzzard.py "<TDI]" -a cc -o ki -stdout | xclip -sel c -noutf8
```
Then `Ctrl + V` in a pcbnew window

(Note this version isn't compatible with the ActionPlugin right now, so just run `git submodule update` from the root of the repo to switch back.)

Demo:
https://user-images.githubusercontent.com/344310/104516849-f0a45b00-5644-11eb-95da-dd3c2894cc90.mp4

pdp7, 2021-01-14:
Yup, it was the issue with Python not refreshing the canvas.  I saved and reopened and it was there.
[image]

pdp7, 2021-01-14:
The command line paste flow worked too
[image]

gregdavill, 2021-01-14:
Awesome! I'll update the readme with this info.

I think until that issue is fixed, and potentially we get a way to invoke the paste command from python, it's probably the best experience using the "Terminal + Paste" Workflow.

closed 2021-01-14 by gregdavill