gregdavill/KiBuzzard #2: AttibuteError: module 'pcbnew' has no atrribute 'Cast_to_FOOTPRINT'

akirasan, 2021-01-13:
Hi,

I tried a simple test, but I have this error. Any idea?

[image]

If I test the buzzard.py (standalone) it works.

[image]

gregdavill, 2021-01-13:
What Version of KiCad are you on? At some point they renamed `MODULES` to `FOOTPRINTS`.
Note: This script has only been tested on KiCad 5.99, Nightly builds.

You might also want to try the workflow outlined in this comment: https://github.com/gregdavill/KiBuzzard/issues/1#issuecomment-759773542

akirasan, 2021-01-15:
KiCad 5.1.9 :/

Obviously this is the problem. I have checked the code for the pcbnew class and it does not have the definition of Cast_to_FOOTPRINT.

https://user-images.githubusercontent.com/3648643/104780355-f30bcf80-5780-11eb-9048-875547cc5904.mp4

closed 2021-01-15 by akirasan

gregdavill, 2021-01-15:
It may be possible to adapt the Buzzard scripts to output v5 compatible footprints. But unfortunately v5 does not handle pasting footprints from the clipboard. So the workflow would be quite different.