gregdavill/KiBuzzard #3: Invoke from plugin icon?

pdp7, 2021-01-14:
Should there be an icon in kicad for it?

@gregdavill seems to:
[image]

But I do not:
[image]

gregdavill, 2021-01-14:
Currently I think the icon I'm getting is some sort of placeholder from KiCad.
Let me add a proper icon.

gregdavill, 2021-01-14:
Can you try c609cd9730f3eef623a6721349c51fa04eaddc66

I've added an icon for it now
[image]

pdp7, 2021-01-15:
@gregdavill I updated and re-launched KiCad but I don't see any icons.  But now that I think about, I don't see any of the other plugin icons like teardrop and RF-Tools.  I'll investigate that as it seems like some issue with my build and/or install of v5.99.  It used to have icons but I don't remember when it went away.

gregdavill, 2021-01-15:
In your pcbnew preferences what does your "Action Plugins" page look like?

[image]

pdp7, 2021-01-16:
Interesting, I don't see KiBuzzard in the list.  Also, not sure why I don't see any of the icons for the plugins that are listed.
[image]

pdp7, 2021-01-16:
In fact, after my most recent build, plugins are listed at all.
I posted about it to see if anyone else having problems.  
https://forum.kicad.info/t/v5-99-plugins-missing-after-recent-build/27046

gregdavill, 2021-01-17:
FYI I'm working on a build I self built 2 days ago.

```
Version: (5.99.0-8406-gb90e72ed07), debug build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh/0.9.3/openssl/zlib nghttp2/1.41.0 librtmp/2.3
Platform: Linux 5.8.0-33-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11
Build Info:
    Date: Jan 16 2021 20:57:04
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OCE: 6.9.1
    Curl: 7.68.0
    ngspice: 32
    Compiler: GCC 10.2.0 with C++ ABI 1014
Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    KICAD_USE_OCE=ON
    KICAD_SPICE=ON
    KICAD_STDLIB_DEBUG=OFF
    KICAD_STDLIB_LIGHT_DEBUG=OFF
    KICAD_SANITIZE=OFF
```

Note that I've got `KICAD_SCRIPTING_WXPYTHON=ON` and `KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON`

jeffwurz, 2021-01-17:
I was able to get it working with fresh install of ubuntu 20.04 in a VM. there were a few issues along the way, perhaps you need to alias python3 to python

sudo ln -s /usr/bin/python3 /usr/bin/python

other than that I think python 3.9 is too new and has other issues

gregdavill, 2021-01-17:
Interesting, thanks! 

On Ubuntu it's probably best to set python with 
```
update-alternatives --install /usr/bin/python python /usr/bin/python3 10
```

pdp7, 2021-01-17:
@gregdavill I actually have `KICAD_SCRIPTING_WXPYTHON=OFF` so that might be the problem.  I changed that recently after a problem trying to run standalone CAM python program for the v5.99 pcbnew module 
https://forum.kicad.info/t/install-the-v5-99-python-modules/26846/11  I will try to get KICAD_SCRIPTING_WXPYTHON=ON working again

closed 2021-02-05 by gregdavill