gregdavill/KiBuzzard #4: Does this support 5.19

VedantParanjape, 2021-01-16:
```
Application: KiCad
Version: 5.1.9, release build
Platform: Linux 5.4.0-59-generic x86_64, 64 bit, Little endian, wxGTK
```

Is this kicad version supported ? I couldn't get it to work, no label was created on pasting

gregdavill, 2021-01-16:
No, currently only the nightly KiCad 5.99 builds are supported.

This is due to better handling of the clipboard in 5.99. 
KiCad 5 unfortunately does not use the system clipboard.

VedantParanjape, 2021-01-16:
Any hacky way to get it working on 5.19 ? like load it as bitmap or something ?

kstubb, 2021-01-16:
Glad I found your comment... spent an hour tonight trying to figure out what I was doing wrong and why it wouldn't show up.

gregdavill, 2021-01-17:
@VedantParanjape I think the only way it would work in v5 is to alter Buzzard directly to output v5 compatible `.kicad_mod` files.

You can see the changes I made here: https://github.com/gregdavill/Buzzard/commit/b59e46e1e7f356e00668a89008306dbe91dad0de

You would have to do something similar so that resulting `.kicad_mod` footprint files were v5 compatible. And then manually insert them like you would a standard footprint.

gregdavill, 2021-01-17:
I have added a note to the README indicating that this currently only supports KiCad 5.99

VedantParanjape, 2021-01-17:
> I have added a note to the README indicating that this currently only supports KiCad 5.99

Cool, will help a lot of devs visiting this page.

VedantParanjape, 2021-01-17:
> @VedantParanjape I think the only way it would work in v5 is to alter Buzzard directly to output v5 compatible `.kicad_mod` files.
> 
> You can see the changes I made here: [gregdavill/Buzzard@b59e46e](https://github.com/gregdavill/Buzzard/commit/b59e46e1e7f356e00668a89008306dbe91dad0de)
> 
> You would have to do something similar so that resulting `.kicad_mod` footprint files were v5 compatible. And then manually insert them like you would a standard footprint.

Cool, this works, yay 🥳

VedantParanjape, 2021-01-17:
> @VedantParanjape I think the only way it would work in v5 is to alter Buzzard directly to output v5 compatible `.kicad_mod` files.
>
> You can see the changes I made here: [...]

I can't find docs for v5 .kicad_mod file format

gregdavill, 2021-01-17:
Pretty sure info is in this document: https://kicad.org/help/legacy_file_format_documentation.pdf

But you can also create a footprint with a silkscreen polygon, and inspect the resulting file. This will probably be the easiest way.
Here is an example:
```
(module Label (layer F.Cu) (tedit 6003E0CC)
  (fp_text reference REF** (at 0 0.5) (layer F.SilkS) hide
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text value Label (at 0 -0.5) (layer F.Fab) hide
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_poly (pts (xy 0.8 1.4) (xy -2.2 0.8) (xy -2.4 -1) (xy -0.2 -1)) (layer F.SilkS) (width 0.1))
)
```
So you'll really just have to populate your own pts with ones generated by buzzard.

VedantParanjape, 2021-01-17:
> Pretty sure info is in this document: https://kicad.org/help/legacy_file_format_documentation.pdf
>
> But you can also create a footprint with a silkscreen polygon, and inspect the resulting file. This [...]

Cool, figured this, what's the tedit string ? seems some kind of hash or timestamp

VedantParanjape, 2021-01-17:
Hi @gregdavill added support for 5.1.x, please merge the PR

https://github.com/gregdavill/Buzzard/pull/1

VedantParanjape, 2021-01-17:
> Glad I found your comment... spent an hour tonight trying to figure out what I was doing wrong and why it wouldn't show up.

https://github.com/VedantParanjape/KiBuzzard, check this out. I'll send a PR soon. This supports 5.1.9

kstubb, 2021-01-17:
> > Glad I found your comment... spent an hour tonight trying to figure out what I was doing wrong and why it wouldn't show up.
> 
> https://github.com/VedantParanjape/KiBuzzard, check this out. I'll send a PR soon. This supports 5.1.9

Nice work, Vedant! I'll give it a go!

Greg;  Thanks for your time as well. I didn't mean to diminish that by any means with my first comment.

I wish I knew what you both know and could help out with stuff like this.

kstubb, 2021-01-17:
I wasn't able to get it to do anything. It installs, it runs -> Create labels. I enter text and hit enter, but I never get anything on the board.

Anyway, I didn't see any way to report issues or discuss it on your github, so I'll wait for that to give full details as I'm sure Greg doesn't want all that here.

Regards;
Kyle

VedantParanjape, 2021-01-17:
> I wasn't able to get it to do anything. It installs, it runs -> Create labels. I enter text and hit enter, but I never get anything on the board.
>
> Anyway, I didn't see any way to report issues or [...]

Hi, Kyle. Github disables issue on forks. Enabled it. Please raise a issue there.

closed 2021-01-19 by VedantParanjape