gregdavill/KiBuzzard #5: Inst err

ok1hra, 2021-01-16:
```
$ pip3 install -r deps/buzzard/requirements.txt --user
Collecting bezier==2020.5.19
  Downloading bezier-2020.5.19.tar.gz (313 kB)
     |████████████████████████████████| 313 kB 1.3 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-yacb9bxn/bezier/setup.py'"'"'; __file__='"'"'/tmp/pip-install-yacb9bxn/bezier/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-q8uziis2
         cwd: /tmp/pip-install-yacb9bxn/bezier/
    Complete output (1 lines):
    The BEZIER_INSTALL_PREFIX environment variable must be set.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

```

gregdavill, 2021-01-17:
I'm not sure I can assist with python setup issues. 

Maybe try following the installation guide from the bezier page? https://github.com/dhermes/bezier/tree/main#installing

jeffwurz, 2021-01-17:
I had success with python 3.8.5, it seems like python3.9 is too new for bezier.

crami, 2021-01-18:
Also have the same issue with python3.9.1

zrecore, 2021-01-25:
Found a workaround. 
[image]

...prepend the `BEZIER_NO_EXTENSION=True` environment variable, and it'll explicitly disable the building of the 'bezier' extension.

```bash
BEZIER_NO_EXTENSION=True pip3 install -r requirements.txt --user
```

iharshadec, 2021-02-02:
> Found a workaround.
> [image]
> 
> ...prepend the `BEZIER_NO_EXTENSION=True` environment variable, and it'll explicitly disable the building of the 'bezier' extension.
> 
> ```shell
> BEZIER_NO_EXTENSION=True pip3 install -r requirements.txt --user
> ```
> 
> See https://bezier.readthedocs.io/en/stable/development.html#building-compiling

I also have 3.9.1 and it's not able to get installed Bezier. OS = WINDOWS 10

BEZIER_NO_EXTENSION=True pip3 install -r C:\Users\Ecloud12\AppData\Roaming\kicad\scripting\plugins\KiBuzzard\requirements.txt --user
'BEZIER_NO_EXTENSION' is not recognized as an internal or external command,
operable program or batch file.

gregdavill, 2021-02-03:
> I also have 3.9.1 and it's not able to get installed Bezier. OS = WINDOWS 10
>
> BEZIER_NO_EXTENSION=True pip3 install -r [...]

On windows you should try to use the command prompt syntax for setting environment variable, rather than the one used by bash.

Which _i think_ is done like this, using the `set` command. (https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/set_1)
```
$ set BEZIER_NO_EXTENSION=True
$ pip3 install -r C:\Users\Ecloud12\AppData\Roaming\kicad\scripting\plugins\KiBuzzard\requirements.txt --user
```

iharshadec, 2021-02-03:
> On windows you should try to use the command prompt syntax for setting environment variable, rather than the one used by bash.
>
> Which _i think_ is done like this, using the `set` command. [...]

Yes, I got bezier installed with this commands. Thank you Greg !

zrecore, 2021-02-04:
Seems this issue can be marked as resolved, and closed.

closed 2021-02-04 by gregdavill