gregdavill/KiBuzzard #7: Characters which go beyound the groundline distort the poligon

crami, 2021-01-19:
I use the fork from @VedantParanjape together with KiCad 5.19 all works great so far, but as it looks like characters like p q and g are causing problems:

[image]

VedantParanjape, 2021-01-19:
I think this is a issue with Buzzard. Not with the plugin. Maybe I am wrong.

gregdavill, 2021-01-19:
Yep. I think in general buzzard was designed for handling uppercase letters. 

How would the best way to handle this, either extend the boarder below the baseline of the text further?

Sodium-Hydrogen, 2021-02-25:
I don't know if you would be interested in integrating a different code base, but I have been working on updating [svg2mod](https://github.com/svg2mod/svg2mod) and I have been working on adding support for svg `<text>` flags and installed type fonts which should be coming soon. Once this is added not only will this be able to automatically support more fonts, but also can use the bounding box to calculate the border. It also has the added advantage of converting from a vector to a vector so paths are sharper.

This is just a thought I had while looking at the KiCAD discord.

arturo182, 2021-02-25:
@Sodium-Hydrogen that does sound interesting, it would also allow multiple lines in a single label or having labels of any height while keeping text size the same.

Sodium-Hydrogen, 2021-03-13:
I have an experimental update on my [personal fork of svg2mod](https://github.com/Sodium-Hydrogen/svg2mod) to add text support and make the tool more useful as a library.

Here is the python code I used to generate the following image.
```
from svg2mod import svg2mod, svg

# mod is the kicad footprint writer class
mod = svg2mod.Svg2ModExportPretty(precision=1)
# t is an svg Text element
t = svg.Text()
mod.add_svg_element(t)

t.set_font(font="DejaVu Sans", italic=False, bold=True)
# Add multiline text
t.add_text("Hello,")
t.add_text("World", origin=svg.Point(0,15))
# This needs to be called to convert raw text to useable path elements
t.convert_to_path()

bbox = t.bbox()
height = bbox[1].y - bbox[0].y
buff = t.size/5
height += buff*2
width = bbox[1].x - bbox[0].x

# Create outline around text with one side being an arc and the other a point
# these are svg path commands lowercase letters mean relative moves and uppercase are absolute moves
pstr = f"M {bbox[0].x},{bbox[0].y-buff} a {height/2},{height/2} 0 0 0 0,{height} h {width} l {height/2},{height/-2} l {height/-2},{height/-2} h {-1*width} z"

p = svg.Path()
p.parse(pstr)
t.paths.append([p])

mod.write()

# Print kicad footprint text created by running mod.write()
print(mod.raw_file_data)
```
[image]

I still need to do some more testing before I push it to release and would appreciate any feedback you might have.

Sodium-Hydrogen, 2021-03-13:
The library will index all fonts found in [common locations](https://github.com/Sodium-Hydrogen/svg2mod/blob/main/svg2mod/svg/svg/svg.py#L891) the first time `svg.Text.load_system_fonts()` is called. After that the tool will keep all found fonts in memory to make future text conversions faster. Also setting `svg.Text.default_font` to the string name of a font will make it so `set_font` doesn't need to be called.

gregdavill, 2021-04-19:
With #39 text can't drop outside of the bounding box.

closed 2021-04-19 by gregdavill