gregdavill/KiBuzzard #10: Pointers not working

cebrax, 2021-01-27:
I am on Windows and everything works great, except the pointers in either ends.

The error message I get when I try to have pointers in both ends or in any of the ends is:
"The system cannot find the file specified."

gregdavill, 2021-01-27:
By pointers, do you mean `<` and `>`?

Can you send a screenshot of the label you're trying to make?

cebrax, 2021-01-27:
Yes, here it is:

[https://i.ibb.co/d4LsFfk/Untitled.png](https://i.ibb.co/d4LsFfk/Untitled.png)

gregdavill, 2021-01-27:
I managed to reproduce this, and it seems like `<` and `>` are used by the windows shell as special characters. They just need to be escaped with a `^` 

Can you try with the latest code?

gregdavill, 2021-01-27:
Whoops, a typo in 32d0224b3f51b08673503970356040d9b81b05f7

But it should be fixed now.
[image]

cebrax, 2021-01-27:
Yup, fixed :) 

Thank you

closed 2021-01-27 by cebrax