gregdavill/KiBuzzard #13: Scale textbox in the new GUI doesn't work

VedantParanjape, 2021-02-02:
The scale input box doesn't seem to work as expected. It doesn't take float input, nor does it set to 0.04 by default.

CC: @arturo182

arturo182, 2021-02-02:
Please provide more info, because it works for me:
[image]

VedantParanjape, 2021-02-02:
Here's the video: https://streamable.com/0ls78m

I am trying to port this to kicad-5.1.9. With your GUI, the command window works fine. If I set scale to anything other than 0, it works fine.

arturo182, 2021-02-02:
Hmm, what if you try changing the 0 [here](https://github.com/gregdavill/KiBuzzard/blob/main/KiBuzzard/dialog/dialog_base.py#L74) to 2, maybe this is less enforced on Windows. If that works, let me know and I'll make a proper fix commit including the wx form file.

VedantParanjape, 2021-02-02:
@arturo182 Yes, this fixed it. But then default value of the text input is 1.0 and not 0.04 (in the code you have set it to 0.04)

VedantParanjape, 2021-02-02:
> Hmm, can you retest with [1a434df](https://github.com/gregdavill/KiBuzzard/commit/1a434df84a6e8f05c83d2fb65a1d81af1b110e15), that should fix the default value too, otherwise maybe try to delete the config.ini and see if after that the default value works. It should be saved between plugin calls.

This worked fine, just one issue.

[image]

I am unable to see the selected brackets in Label drop box

DCelectronics, 2021-02-02:
I have the same issue with KiCad v5.1.9 on Linux only (Win10 is OK).
I implemented fix https://github.com/gregdavill/KiBuzzard/commit/1a434df84a6e8f05c83d2fb65a1d81af1b110e15 but I changed it to 2 decimal places for consistency with the default behaviour in Win10 (besides 0.001 is too fine step).

gregdavill, 2021-02-03:
It might be nice to have the scale setup in nicer units, like mm or mils.

Right now it's a bit disconnected, is it supposed to be inches? Buzzard doesn't really make this clear.

DCelectronics, 2021-02-03:
I believe it is in inches for the text height.

closed 2021-02-11 by VedantParanjape