gregdavill/KiBuzzard #14: TypeError: in method 'new_Dialog', expected argument 1 of type 'wxWindow *'

Plasmabot1, 2021-02-08:
Hello, I encounter this error when launching the plugin by clicking the icon in pcbnew.

Versions:
Windows 10
Kicad 5.99-r20791.34d7c3e31f (2021-02-08)
Latest KiBuzzard (just now cloned)

```
Traceback (most recent call last): 
File "C:\Users\Anhad\AppData\Roaming\kicad\5.99\s...\plugins/KiBuzzard/KiBuzzard/__init__.py", line 112, in Run 
dlg = Dialog(self,pcbnew_frame, self.config, self.buzzard_path, run buzzard) 

File "C:\Users\Anhad\AppData\Roaming\kicad\5.99\s...\plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 11, in __init__ 
dialog_base.KiBuzzardDialog._init_(self, parent) 

File "C:\Users\Anhad\AppData\Roaming\kicad\5.99\s...\plugins/KiBuzzard/KiBuzzard/dialog/dialog_base.py", line 20, in __init__ 
wx.Dialog._init_ ( self, parent, id = wx.ID_ANY, title = u"KiBuzzard", pos = wx.DefaultPosition, size = wx.Size( 420,280 ), 
style = wx.CLOSE_BOX|wx.DEFAULT_DIALOG_STYLE|wx.RESIZE_BORDER ) 

File "C:\Program Files\KiCad\lib\python2.7\site-packages\wx-3.0-msw/wx/_windows.py", line 734, in __init__ 
_windows_.Dialog_swiginit(self,_windows_.new_Dialog(*args, **kwargs)) 

TypeError: in method 'new_Dialog', expected argument 1 of type 'wxWindow *'
```

Thank you for your help

Plasmabot1, 2021-02-08:
Update: 
The problem seems to have resolved itself now. The dialog window opens, but I have a separate problem. I have opened a new issue for that.

Plasmabot1, 2021-02-08:
Update 2:
It works sometimes and it doesnt work other times. I have no idea why. After relaunching kicad it returned.

gregdavill, 2021-04-11:
Can you try the latest version, this may be resolved now? 
Feel free to open up a new issue if you're having issues.

closed 2021-04-11 by gregdavill