gregdavill/KiBuzzard #15: The system cannot find the path specified

Plasmabot1, 2021-02-08:
Hello, I have this error:

No matter what settings I put, I get this.
[image]

Versions:
Windows 10
Kicad 5.99-r20791.34d7c3e31f (2021-02-08)
Latest KiBuzzard (just now cloned)

All dependencies should be satisfied, I successfully ran `pip3 install -r requirements.txt --user` after cloning.

Thank you for your help.

gregdavill, 2021-02-08:
I think the error message could be better. I suspect it's unable to find a local install of python.

I'm thinking the error is coming from here: https://github.com/gregdavill/KiBuzzard/blob/1a434df84a6e8f05c83d2fb65a1d81af1b110e15/KiBuzzard/__init__.py#L88

Can you try installing Python 3.8 to `C:/Python38`
https://www.python.org/downloads/release/python-387/

Plasmabot1, 2021-02-08:
Ok it works now. 
However, I would much rather not have yet one more python installation. Can you make it use whatever python is there in PATH?

arturo182, 2021-02-09:
No, because the one in PATH first is the KiCad one that's 2.7 and won't work.

As for the message, we can't change the text because it comes from the system, but we could probably preempt it by checking if the executable exists and showing our own message first.

Plasmabot1, 2021-02-09:
I have python 3.9 installed and in PATH and that is what comes up when I type `python` in cmd. Do plugins use different environment variables?

gregdavill, 2021-02-09:
It gets a little bit complicated, I think we could probably use the python from PATH.

Plasmabot1, 2021-02-09:
Python's default install location is `C:\Users\XXXXX\AppData\Local\Programs\Python\PythonXX\` if that helps

Plasmabot1, 2021-02-09:
What lines to I have to change in order to direct it to my current python installation?

gregdavill, 2021-02-09:
This line: https://github.com/gregdavill/KiBuzzard/blob/1a434df84a6e8f05c83d2fb65a1d81af1b110e15/KiBuzzard/__init__.py#L88

If you just change it to
```python
process = subprocess.Popen(['python.exe', buzzard_script] + args, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)
```
it should be running the python version from your path.

Plasmabot1, 2021-02-09:
Ok thank you. Feel free to close this if you don't think it needs changing.

Plasmabot1, 2021-02-09:
#

closed 2021-02-09 by Plasmabot1

DCelectronics, 2021-02-13:
@Plasmabot1 I have found two users that had issue with making KiBuzzard work with Python 3.9. They were able to run KiBuzzard after installing Python 3.8 and setting the user Environment Variables.
Also, please see https://github.com/gregdavill/KiBuzzard/issues/18#issuecomment-778667973

Plasmabot1, 2021-02-13:
Yeah, it seems everyone has a different way that Python is set up. My solution works for me, user environment variables works for you, etc. I don't know if there is one method that will work for everyone. The ideal solution is for it to use the python3 that is eventually going to be bundled with KiCad on windows, but that hasn't happened yet.
Until then, maybe the readme should just tell people to point it to the python3 location as an installation step?

gregdavill, 2021-02-13:
I'm already investigating making use of the internal python on KiCad. 
The main hurdles are 
 - v5 still uses 2.7. Which we'd need to support
 - Currently v5.99 windows is also packaged with 2.7.
 - Installing extra packages for the bundled python is tricky on Windows.

I think this is a better path to work on that to keep trying to mess around with paths to separate python installs.

It's not 'functional' yet. But you can check out this newer approach here: https://github.com/gregdavill/KiBuzzard/tree/dev
I've got the most complicated part working, text rendering to polygons without needing Bezier,Numpy. The other dependencies are packaged directly with the plugin.