gregdavill/KiBuzzard #18: [Soltution] How to make it run on windows

elekgeek, 2021-02-13:
The message "Can't find the path specified" is related to python.exe path error in `__init__.py` inside the second KiBuzzard directory i.e. `plugins\KiBuzzard\KiBuzzard\__init__.py`

 Scroll down to line 88:

`process = subprocess.Popen(['C:\\Python38\\python.exe', buzzard_script] + args, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)`

and change it to:

`process = subprocess.Popen([buzzard_script] + args, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)`

Also note that you need Python 3.8.3 because you will be able to install all dependencies found inside requirements.txt. First clone the plugin recursively to make sure all required files are downloaded, and then open a cmd window inside plugins\KiBuzzard directory then issue the command:
`pip3 install -r requirements.txt --user`

elekgeek, 2021-02-13:
This is just a quick solution..

closed 2021-02-13 by elekgeek

DCelectronics, 2021-02-13:
@elekgeek 
Since some users may install python 3.8 for a single user instead of system the path may not always be C:\\Python38\\python.exe.
My proposed approach is to have defined the path to python 3 through user Environment Variables. 
This link explains how to set python3 user Environment Variables.
https://datascience.com.co/how-to-install-python-2-7-and-3-6-in-windows-10-add-python-path-281e7eae62a
By doing this, I have discovered that the code becomes more portable between various Python installation settings on Windows, and the line:
```process = subprocess.Popen(['C:\\Python38\\python.exe', buzzard_script] + args, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)```
can be revised as:
```process = subprocess.Popen(['python3', buzzard_script] + args, stdout=subprocess.PIPE, stderr=subprocess.PIPE, shell=True)```