gregdavill/KiBuzzard #19: Index out of range error

reportingsjr, 2021-02-20:
I've just cloned this repo into the 5.99 plugins directory using the command in the readme.

When clicking the button to launch the plugin in pcbnew, I get the following error:
[image]

This using the fedora nightly, updated just an hour ago.

Kicad build settings:
[image]

Python version info:
[jon@arikem dev]$ python3 --version
Python 3.8.7

gregdavill, 2021-02-20:
Can you open a scripting console in pcbnew and run this:

```python
>>> import wx
>>> [print(x.GetTitle()) for x in wx.GetTopLevelWindows()]
```

For reference I get the following:
```
>>> [print(x.GetTitle()) for x in wx.GetTopLevelWindows()]
meiji-led-ring — KiCad 5.99

meiji-led-ring — Pcbnew

KiCAD:PCBNEW - Python Shell - PyAlaMode 0.9.8
[None, None, None, None, None]
```

reportingsjr, 2021-02-21:
`[GCC 10.2.1 20201125 (Red Hat 10.2.1-9)] on linux
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/jon/.config/kicad/5.99/PyShell_pcbnew_startup.py

import wx
[print(x.GetTitle()) for x in wx.GetTopLevelWindows()]
obsidian-boa-r0.1 — KiCad 5.99
obsidian-boa-r0.1 — PCB Editor

KiCAD:PCBNEW - Python Shell - PyAlaMode 0.9.8
[None, None, None, None]
`

gregdavill, 2021-02-21:
Interesting your pcbnew window comes with the title `PCB Editor` 

Easy enough to add that to the check: https://github.com/gregdavill/KiBuzzard/commit/e9f754c75a3555868efbd4b453abd4b32fbaa6c2

arturo182, 2021-02-23:
I just updated my nightly and ran into the exact same issue, guess they changed the title recently.

arturo182, 2021-02-23:
Here's the change 4 days ago: https://gitlab.com/kicad/code/kicad/-/commit/66b33834b5f5f44150080aed645037ebd9f3eee3#83a257d25158f354cabcb709bdd360a2778ca729_173_173

reportingsjr, 2021-02-23:
Yep, that fixed it! Thanks Greg!

closed 2021-02-23 by reportingsjr