gregdavill/KiBuzzard #20: Python error 'init_fs_encoding' on macOS

robin7331, 2021-02-25:
After successfully installing the plugin I am able to open the modal window from the PCB Editor.
As soon as I hit `Create!` or `Enter` to create the actual label I am faced with this python error 👇

[image]

After confirming I get a second dialog 

[image]

I suspect its because KiCad 5.99 comes with Python 2.7 builtin. Maybe `init_fs_encoding` is not supported in 2.7? Not sure.. Any idea? 🤷‍♂️

iharshadec, 2021-02-25:
@robin7331 try Greg's this branch:- https://github.com/gregdavill/KiBuzzard/tree/dev

gregdavill, 2021-02-25:
The `dev` branch is still a WIP, but it's trying to make use of the internal python directly. And has been tested on atleast 1 OSX system. You will need to install freetype, which is recommended through macports/brew.

Unfortunately the UI is lacking, but if you're comfortable tweaking the python code directly it can do the same things.

robin7331, 2021-03-05:
I have tried the `dev` branch and it seems like I can now go one step further.
But it has a weird font and it does not seem to fill the background.
Also when I click `Create` nothing really happens. 

> You will need to install freetype, which is recommended through macports/brew.

This was already installed. I even reinstalled it. 
No luck with `dev` or or `main` for me so far 🤷‍♂️

[image]

gregdavill, 2021-03-06:
Yep, I'm trying out some different fonts. The breaks in the background are more an issue for the preview, and aren't apparent when manufactured.

With the dev branch, after pressing create does it close? 
What happens if you try to paste Cmd+V into KiCad?

The plugin is supposed to copy a footprint to the clipboard, then paste it onto the KiCad window, this paste step might now work on MacOS, as it's requires us requesting a different window gain the UI focus.
If the auto-paste fails you should still be able to manually paste.

If that doesn't work can you confirm that the footprint is making it to your clipboard by trying to paste into a text editior?

CaptClaude, 2021-03-09:
I tried the man branch here as well as the main branch from @VedantParanjape without success. Vedant helped me by showing how to see the error that is thrown (open from terminal) which was something of a revelation.
Installed the dev branch and was able to create my first labels.  They still locate up at (0,0) but that's a pretty minor issue at this point.  Here is what I see:
[image]
[image]
[image]
Clearly we are moving in the right direction (for me at least).
@gregdavill to answer your question, after clicking Create, the label appears and can be dragged around. Cmd+V does nothing and checking with a text editor indicates the paste buffer is empty.

If there are other things you'd like me to try, let me know.

CaptClaude, 2021-03-10:
Some additional info gathered running KiCAD from a command line (OS X):
```
Bardolino:plugins jm$ /Applications/Kicad/kicad.app/Contents/MacOS/kicad
```
KiCAD app window opens.
Next, click on PCBNew button:
```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
/Applications/Kicad/kicad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa/wx/_core.py:16633: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
```
Click KiBuzzard button, dialog opens, paste 6 characters into field:
```
WARN: No Position Table found for this typeface. Composition will be haphazard at best.
create svg
SVG width not found, guessing based on scale factor
Translating Path 1 of 6
Translating Path 2 of 6
Translating Path 3 of 6
Translating Path 4 of 6
Translating Path 5 of 6
Translating Path 6 of 6
WARN: No Position Table found for this typeface. Composition will be haphazard at best.
create svg
SVG width not found, guessing based on scale factor
Translating Path 1 of 6
Translating Path 2 of 6
Translating Path 3 of 6
Translating Path 4 of 6
Translating Path 5 of 6
Translating Path 6 of 6
```
Click create:
```
/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/tmpAulhVO/buzzard_labels.pretty
```

After setting 
```
export PYTHONHOME=/Library/Frameworks/Python.framework/Versions/3.7:/Library/Frameworks/Python.framework/Versions/3.7/bin
```
The warning about platform independent libraries went away.
Don't know how to deal with this:
```
WARN: No Position Table found for this typeface. Composition will be haphazard at best.
create svg
SVG width not found, guessing based on scale factor
Translating Path 1 of 1
```

gregdavill, 2021-04-11:
Can you try the latest version, this may be resolved now? 
Feel free to open up a new issue if you're having issues.

closed 2021-04-11 by gregdavill

robin7331, 2021-04-11:
I am still getting an error when using the version from the main branch. 
But I'll investigate and create a new issue if necessary. 👋

[image]

gregdavill, 2021-04-11:
That's unrelated, and related to this: https://github.com/wxFormBuilder/wxFormBuilder/issues/327

I'll fix-up my wxFormBuilder version and I should be able to fix that.

gregdavill, 2021-04-11:
@robin7331 Is that fixed in bcbd282e1f5ccadeb98415857928c482db0fc544?

robin7331, 2021-04-11:
Jep that helps. I am hacking around as well at the moment and there seem to be many other issues with `wx`. 

Had to call `wx.StockGDI._initStockObjects()` to initialize all the `wx.{{color}}` calls. 

Next, I had to remove all the `SetToolTip` calls because of : 
```
File "/Users/robin/Library/Preferences/kicad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog_text_base.py", line 69, in __init__
self.m_MultiLineText.SetToolTip( _(u"Enter the text placed on selected layer.") )

File "/private/var/folders/0_/50zzr6qs1v90gkw4sg2qkfph0000gn/T/AppTranslocation/F288DBE0-034A-47DA-BC08-B40B1802B0B5/d/KiCad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 11394, in SetToolTip
return _core_.Window_SetToolTip(*args, **kwargs)

TypeError: in method 
```

It's still not working though. By the way. What is your workflow? When I edit the code I have to restart KiCad in order to test it. When I hit 'Refresh Plugins' the KiBuzzard icon disappears. 
[image]

robin7331, 2021-04-11:
Created a new issue and partially copied stuff from here to the new issue.
Just to keep things clean. 

#27