gregdavill/KiBuzzard #21: Label with space and angle bracket does not show correct

ifadiga, 2021-02-28:
Hi,
I use KiBuzzard plugin with KiCad 5.99 nightly and have a problem when using the angle bracket, with space before or after the text in the label.
In the picture is shown that I get arrow character before or after the text when using space and sometimes label not show correct (created a square polygon, instead of angle polygon on start). 

[image]

putyn, 2021-03-11:
@ifadiga you are most likely using windows, right ? I have a pull request #24 that will hopefully fix it 😄 just waiting for it to be merged.

ifadiga, 2021-03-15:
@putyn thanks for the reply.
Yes, I am using Windows.
Ok, I had fixed it to label what I need.

gregdavill, 2021-04-11:
Can you try the latest version? I have a feeling this is resolved now with the new UI changes.
Feel free to open up a new issue if you're having issues.

closed 2021-04-11 by gregdavill