gregdavill/KiBuzzard #22: Question & Volunteer

CaptClaude, 2021-03-02:
I want to use KiBuzzard on the production version of KiCAD on a Mac (5.1.6).  In spite of it only being guaranteed to work on 5.99, @arturo182 has said it works for those not on the bleeding edge as well, so I will give it a shot.  To move myself in the right direction, I installed Buzzard (not knowing any better). To do that I installed the latest version of Python. Buzzard runs without faceplanting.  Now I see that I should 
> Install the script in your KiCad scripting directory

As Noob as it might seem (and is), I cannot figure out where the _scripting directory_ is under MacOS.  My Google-Fu is weak.
Question: Where is the scripting directory under MacOS?
If I succeed in getting it installed, and it runs, I would be happy to write up my steps for posterity and to test as new releases come are made available.

gregdavill, 2021-03-02:
I know that @flummer has had some success with it working on OSX.

flummer, 2021-03-02:
@CaptClaude I'm pretty sure the plugins go in `~/Library/Application Support/kicad/scripting/plugins` (so the Library folder in your user dir) or maybe `~/Library/Preferences/kicad/scripting/plugins` when we are talking about the stable 5.1.x.

I can't remember if it's also an option in the stable version, but in nightly in pcbnew, there is "Tools" -> "External Plugins" -> "Reveal Plugin Folder in Finder". In nightly I actually think both the above path (one of them) AND the new versioned path under the "Documents" folder is searched for plugins.

I have only tried KiBuzzard in nightly, and gotten it to work with the `dev` branch of this repo.

If you run the following two commands in the python shell inside pcbnew, I think you will get the paths:

```
import pcbnew
print(pcbnew.NOT_LOADED_WIZARDS)
```

CaptClaude, 2021-03-03:
OK, we continue to move in the right direction. Very late last night I figured out that the scripting directory should be in `~/Library/Preferences/kicad` but when I looked, there was nothing there. So I created `/scripting/plugins` and ran `git clone https://github.com/gregdavill/KiBuzzard --recursive`, which seemed to work just fine.  Then I cd'd into the newly created KiBuzzard directory and ran `pip3 install -r requirements.txt --user` -- that's when things faceplanted.

> Bardolino:KiBuzzard jm$ pip3 install -r requirements.txt --user
> Collecting bezier==2020.5.19
>   Downloading bezier-2020.5.19.tar.gz (313 kB)
>      |████████████████████████████████| 313 kB 2.9 MB/s 
>     ERROR: Command errored out with exit status 1:
>      command: /Library/Frameworks/Python.framework/Versions/3.9/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/pip-install-n8fy5f4p/bezier_7092cba881e84db4b2e6d1bef21f8136/setup.py'"'"'; __file__='"'"'/private/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/pip-install-n8fy5f4p/bezier_7092cba881e84db4b2e6d1bef21f8136/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/pip-pip-egg-info-33a5cwx0
>          cwd: /private/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/pip-install-n8fy5f4p/bezier_7092cba881e84db4b2e6d1bef21f8136/
>     Complete output (1 lines):
**>     The BEZIER_INSTALL_PREFIX environment variable must be set.**
>     ----------------------------------------
> WARNING: Discarding https://files.pythonhosted.org/packages/b9/0c/b1982f93c36fc06850a3880bb693bf24a047946633ca7403664836b9da9e/bezier-2020.5.19.tar.gz#sha256=ff75337e5bb4385376aeaa48fc941bcd61b528df43a1e938bfb07fe7d924b19c (from https://pypi.org/simple/bezier/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
> ERROR: Could not find a version that satisfies the requirement bezier==2020.5.19
> ERROR: No matching distribution found for bezier==2020.5.19
>  

I then spent time trying to figure out what `The BEZIER_INSTALL_PREFIX environment variable must be set.` and realized I was in _waaaay_ over my head and gave up.
I also tried various ways of installing `xclip` as well, without success.

That said: Upon reading what @flummer wrote, and with PCBNew running and a PCB layout on the screen, Tools -> Refresh Plugins, then Tools -> Create Labels and there was a dialog box open!
[image]

[image]
In extreme Noob fashion, I am again stuck as I don't know what comes next. I'm farther along than I was yesterday.

Also, in the sable version, there is no `"Tools" -> "External Plugins" -> "Reveal Plugin Folder in Finder"`. I assume that some folders will be moving around when 5.99 changes to 6.X.

Thanks.

arturo182, 2021-03-03:
See #5 for the Bezier issue

CaptClaude, 2021-03-04:
Note to self: Stay off the bleeding edge of Python. I had 3.9.2 (plus the Apple 2.x) and Numpy is not up to speed with 3.9.2. After googling a bit, I installed 3.7.9 and `pip3 install -r requirements.txt --user` ran without a hitch. Now I have a label icon on the toolbar. Do I need the MacOS equivalent of `xclip`?

When I run KiBuzzard, I get the dialog box above and when I click Create! the dialog box becomes modal and every menu in KiCAD grays out. I can switch between GUI and Command Line but that's about it until I close the KiBizzard dialog box.

Am I missing something or is this expected when using KiBizzard in the production version?  

Thanks.

gregdavill, 2021-04-11:
Can you try the latest version? 
Feel free to open up a new issue if you're having issues.

closed 2021-04-11 by gregdavill

CaptClaude, 2021-04-14:
The version current as of 4/13/2021 works as advertised. Problems solved!
Thanks.