gregdavill/KiBuzzard #25: ~ in text crashes KiCad

pboe, 2021-03-11:
Hi,
awesome tool, thanks for your work.

1. open PCB Editor
2. Create new label with gui
3. enter ~, the character is highlighted until you add another character like  n or space
4. If you click create while the ~ is highlighted, KiCad crashes
5. if if youo press ~ and space everything is fine

Latest master KiBuzzard 
and 
Application: KiCad

Version: (5.99.0-9683-g3db8da4be4), release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.39.2

Platform: macOS Catalina Version 10.15.7 (Build 19H524), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar  8 2021 19:25:55
	wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
	Boost: 1.75.0
	OCC: 7.5.0
	Curl: 7.54.0
	ngspice: 31
	Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

gregdavill, 2021-04-11:
I think this might be fixed now. Can you confirm?

https://user-images.githubusercontent.com/344310/114296274-871fd580-9ae9-11eb-862b-617c7c25d93f.mp4

pboe, 2021-04-12:
Unfortunately not.

Using latest KiCad nightly (5.99.0-10229-ge8e7d13299)

executed git pull /Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard

  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/__init__.py", line 1, in <module>
    from . import KiBuzzard
  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/__init__.py", line 46, in <module>
    from .plugin import KiBuzzardPlugin
  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/plugin.py", line 12, in <module>
    from .buzzard.buzzard import Buzzard
  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/buzzard/buzzard.py", line 5, in <module>
    from freetype import Face
  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/buzzard/../deps/freetype/__init__.py", line 22, in <module>
    from freetype.raw import *
  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/buzzard/../deps/freetype/raw.py", line 52, in <module>
    raise RuntimeError('Freetype library not found')
RuntimeError: Freetype library not found

It's still working using my old version by going back: git reset --hard c3bf89f6e45dc7a9659601e792284eb87ff2529b

gregdavill, 2021-04-12:
You will need to install the freetype libraries.
There is a mention of this is the readme: https://github.com/gregdavill/KiBuzzard#installation

I think using `pip` will work and is the easiest way. But if there is better instructions I can add for MacOS in particular, let me know and I'll add them.

robin7331, 2021-04-12:
I think I installed freetype via `pip3 install freetype-py`. Note the `pip3` instead of `pip`.
But I am not sure if that's because I installed other versions of python on my system that others may not have.
 
Python is a big mystery to me because I never actually know which version is being used by a command/tool/script.
Also I do not know why it's okay to install freetype with my local pip3 while KiCad is running its own python version internally.  🤷‍♂️

[image]

Also I don't even have `pip` but only `pip3`

[image]

robin7331, 2021-04-12:
@pboe what keyboard and input source are you using? 
You find the keyboard source in your macOS preferences under `Keyboard > Input Sourcees`

robin7331, 2021-04-12:
Just an idea. I am having a QWERTY keyboard with a US Input Source. 
To type an `~` I need to hit **Shift + `** release and then hit space for it to become a **~**
I assume something crashes along those lines. Since its more than a simple keypress.
Maybe Buzzard/FreeType is rendering some in between stuff that's actually only suppose to happen when the full keystroke combination was entered. 

But I cannot reproduce this error on my Mac.. 

https://user-images.githubusercontent.com/3868463/114368394-aa946000-9b7d-11eb-849b-5187cf8a250d.mp4

robin7331, 2021-04-12:
Oh man never mind. I did not see the comment with the `FreeType not found` error.
So yes.. that's probably the only reason why it crashes. :D

gregdavill, 2021-04-12:
I suspect the `~` was causing an issue with the older version because the string was passed as an argument to `python Buzzard.py` 
I'm pretty sure it's now just a freetype issue, I was just pinging you because I know you were on MacOS too, and got it working :)

Installing `freetype-py` should work, since part of that installation process it installs the freetype libraries system-wide.

pboe, 2021-04-12:
It's a mess  with all the python version on macOS
I installed freetype on my old version with the following command:
/opt/local/Library/Frameworks/Python.framework/Versions/3.8/bin/pip3 install -r requirements.txt --user
checking now 
/opt/local/Library/Frameworks/Python.framework/Versions/3.8/bin/pip3 install freetype-py --user
results in:
Requirement already satisfied: freetype-py in /Users/jo/Library/Python/3.8/lib/python/site-packages (2.1.0.post1)

So for me it looks like freetype is installed. Going back to the old version is still working:
`It's still working using my old version by going back: git reset --hard c3bf89f`

gregdavill, 2021-04-12:
Do you use homebrew? Maybe installing via homebrew will work? https://formulae.brew.sh/formula/freetype

It might be that installing via `--user` doesn't correctly install system-wide libraries.

Yep, there has bees some big UI/structural changes. So it's not surprising that an old commit is working.

pboe, 2021-04-12:
Installed freetype with macports but no success.

I've never been into python, so i have absolutely no idea how all those dependencies and modules are managed.
The new KiBuzzard version  added a directory KiBuzzard/KiBuzzard/deps/freetype/ from which the script executes raw.py and fails.
If i rename  KiBuzzard/KiBuzzard/deps/freetype KiBuzzard is working and for me it looks like it is using the system version of freetype.

pboe, 2021-04-12:
Back to the original bug:
With the actual nightly KiCad doesn't crash anymore with the old version, but works as expected. ~without a folowing space still beeing highlighted is accepted and can be pasted into the pcb.

The new version behaves different:
Entering ~character doesn't show any character in the text field. That's not a normal behavior on macOS.

Entering ~ followed by a space character shows the ~ in the text field and it's previewed and pasted correctly into the PCB

Entering ~ followed by character n results in the ñ character in the text field but nothing in the preview.
Deleting the ñ character with delete and  type any text still leaved the preview empty

[image]
[image]

gregdavill, 2021-04-12:
Ahh! I think I see what you mean.

So you're using a special input method? So when you type '~' it's not actually typing `~` until you enter another letter `u` to create `ũ`? 

I suspect maybe while the `~` is selected it's not actually entered into the text box.. 
So maybe it's an empty string is passing through and causing the crash.

robin7331, 2021-04-12:
That's exactly what I had suspected earlier with the video demonstration 😅
Stil thin its a different issue..

gregdavill, 2021-04-12:
It may be that `FredokaOne` may not support accents. 
But we should catch this and show an error.

gregdavill, 2021-04-12:
I get this error when I paste this `ũee` and I have `FredokaOne`

[image]

If I use `UbuntuMono-B` I do get a preview

[image]

For me the error hasn't crashed anything, once I dismiss the error, things work as expected.

pboe, 2021-04-12:
> Ahh! I think I see what you mean.
> 
> So you're using a special input method? So when you type '~' it's not actually typing `~` until you enter another letter `u` to create `ũ`?
Correct, it's exactly how you enter characters like â or é. On the Mac you can do this as well for ñ & õ.
Its the same behavior for all the other fonts, so maybe it's not font related.
Once i entered ñ, even if i delete it with backspace, the formular stops working and showing any characters in the preview field
The other issue, why i have to delete  KiBuzzard/KiBuzzard/deps/freetype stays.

Thanks for Your efforts.

robin7331, 2021-04-12:
If I try to create a label with `ũee` and `FredokaOne` I get this error in the log file.
(that was created with the code from my PR #33 )

```
04-12 13:49:32 STDERR 142:Traceback (most recent call last):
04-12 13:49:32 STDERR 142:  File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 89, in labelEditOnText
04-12 13:49:32 STDERR 142:while self.sourceText != self.CurrentSettings():
04-12 13:49:32 STDERR 142:  File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 85, in CurrentSettings
04-12 13:49:32 STDERR 142:self.m_JustifyChoice1.GetStringSelection() + self.m_JustifyChoice.GetStringSelection()
04-12 13:49:32 STDERR 142:UnicodeEncodeError
04-12 13:49:32 STDERR 142::
04-12 13:49:32 STDERR 142:'ascii' codec can't encode character u'\u0169' in position 0: ordinal not in range(128)
04-12 13:49:33 STDERR 142:Traceback (most recent call last):
04-12 13:49:33 STDERR 142:  File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 173, in OnOkClick
04-12 13:49:33 STDERR 142:self.func(self, self.buzzard)
04-12 13:49:33 STDERR 142:  File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/plugin.py", line 81, in run_buzzard
04-12 13:49:33 STDERR 142:footprint_string = p_buzzard.create_v6_footprint()
04-12 13:49:33 STDERR 142:  File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/buzzard/buzzard.py", line 503, in create_v6_footprint
04-12 13:49:33 STDERR 142:for poly in self.polys:
04-12 13:49:33 STDERR 142:AttributeError
04-12 13:49:33 STDERR 142::
04-12 13:49:33 STDERR 142:Buzzard instance has no attribute 'polys'
```

pboe, 2021-04-19:
With the cleaned up dependencies, the actual master is now working out of the box.
Unfortunately german umlauts öäü now also break the plugin.
Once an umlaut is entered the plugin stops working, even if you delete the umlaut. Deleting the character, closing and reopening the plugin restores functionality.

gregdavill, 2021-04-19:
Do you have a `kibuzzard.log` or `kibuzzard/error.log` to share? These should be generated when you've run the plugin.

pboe, 2021-04-19:
[kibuzzard.log](https://github.com/gregdavill/KiBuzzard/files/6334834/kibuzzard.log)

gregdavill, 2021-04-19:
So your log has this error:

```
04-19 10:54:59 STDERR 138:Traceback (most recent call last):
04-19 10:54:59 STDERR 138:  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/dialog/dialog.py", line 97, in labelEditOnText
04-19 10:54:59 STDERR 138:while self.sourceText != self.CurrentSettings():
04-19 10:54:59 STDERR 138:  File "/Users/jo/Documents/KiCad/5.99/scripting/KiBuzzard/KiBuzzard/dialog/dialog.py", line 94, in CurrentSettings
04-19 10:54:59 STDERR 138:self.m_JustifyChoice1.GetStringSelection() + self.m_JustifyChoice.GetStringSelection()
04-19 10:54:59 STDERR 138:UnicodeEncodeError
04-19 10:54:59 STDERR 138::
04-19 10:54:59 STDERR 138:'ascii' codec can't encode character u'\xf6' in position 0: ordinal not in range(128)
```

I've switched to handling these settings directly as utf-8 objects, instead of trying to convert to 'ascii'. 
Maybe this will fix the issue?

pboe, 2021-04-19:
Sounds reasonable

pboe, 2021-04-19:
Bug fixed. Thanks for the support and all the effort you put into that plugin.
That's very well appreciated.

gregdavill, 2021-04-19:
Awesome! 
So it's all working now? I'll close this issue then :)

closed 2021-04-19 by gregdavill