gregdavill/KiBuzzard #26: can't move label to back silkscreen layer

agilum, 2021-04-07:
hi Greg, 

With the recent kicad-nighly update I can not move a label to B.Silkscreen.  Label's properties changed to B.Silkscreen, but label stays on F.Silkcreen and text is unaltered. All  was  fine several updates before.

Can You help?

A.D.

I am using :

 Application: KiCad PCB Editor

Version: 5.99.0-unknown-b2f788adc1~125~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.10.0 x86_64, 64 bit, Little endian, wxGTK, i3, x11

Build Info:
	Date: Apr  6 2021 08:04:15
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.3.0
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

gregdavill, 2021-04-10:
Can you try this?

Place a label on the top silkscreen. 
Then use the `F` key to flip it.

https://user-images.githubusercontent.com/344310/114269920-a581c480-9a48-11eb-88a5-02c75da24018.mp4

closed 2021-04-10 by agilum

agilum, 2021-04-10:
Thanks a lot. Verified. Works great.