gregdavill/KiBuzzard #27: wx related errors when using KiCad 5.99 nightly macOS

robin7331, 2021-04-11:
When running KiBuzzard on the KiCad 5.99 nightly macOS build I get a lot of `wx` related issues. 
They all seem to occur from the `dialog_text_base.py` file which is auto-generated by the `wxFormBuilder`.

**_Afaik this KiCad version ships with wx version 3.0._**

Had to call `wx.StockGDI._initStockObjects()` to initialize all the `wx.{{color}}` calls before using them. 

Next, I had to remove all the `SetToolTip` calls because of : 
```
File "/Users/robin/Library/Preferences/kicad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog_text_base.py", line 69, in __init__
self.m_MultiLineText.SetToolTip( _(u"Enter the text placed on selected layer.") )

File "/private/var/folders/0_/50zzr6qs1v90gkw4sg2qkfph0000gn/T/AppTranslocation/F288DBE0-034A-47DA-BC08-B40B1802B0B5/d/KiCad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 11394, in SetToolTip
return _core_.Window_SetToolTip(*args, **kwargs)

TypeError: in method 
```

It's still not working though and I suspect its a wx version incompatibility issue.

gregdavill, 2021-04-11:
I think there are some troubles with wx versions across KiCad 5.1 and 5.99 and Win/OSX/Linux.

You're in 5.99 right? if you refresh via Prefs->Action Plugins then you can see a log if you hit any errors.

I generally just run `python KiBuzzard/test_dialog.py' to test UI changes.

robin7331, 2021-04-11:
Yes I am on 5.99. 

[image]

When I refresh the Action Plugins KiBuzzard simply disappears from the list. 
No error dialog or logs in the scripting console. 🤔

robin7331, 2021-04-11:
Did you fix the `SetSizeHintsSz` directly in the file or by changing some property in your `wxFormBuilder` ?

gregdavill, 2021-04-11:
Yes, I did that one directly. 

Ideally it would be fixed in wxFormBuilder. But the version that supports wx3.0 pre-dates github, and it on sourceforge.

robin7331, 2021-04-11:
The thing is how to deal with this incompatibility? 
(re)do it entirely in wx3.0? 
Is that even compatible with newer versions? Or is it then working on macOS KiCad but incompatible with others?

robin7331, 2021-04-11:
The funny thing is. I think `SetSizeHintsSz` is not even a implemented in wx 3.0.4.
At least I cannot find it. It's not on the `wxDialog` or any of its super classes 🤷‍♂️

robin7331, 2021-04-11:
I managed to get the dialog open. But I do not see a preview.
[image]

If I paste something onto the canvas after hitting 'ok' I only get this cross.
[image]

It seems to be empty of some sort
[image]

Somethings missing...

gregdavill, 2021-04-11:
Can you run KiCad from a terminal to see it's output? Might let me know if you're hitting an error.

robin7331, 2021-04-11:
When launching from the terminal I can't even open the `*.kicad_pcb` file. 
I get this error:

[image]

But that's not really KiBuzzard related I guess.

gregdavill, 2021-04-11:
Can you try the latest commit? I'm not catching the exceptions now. 
So if you hit one it should show a popup I think.

gregdavill, 2021-04-11:
Latest commit will output a traceback if there is an exception in generating the preview.

The "preview" actually generates the polys needed by the footprint, so it's not surprising that your footprint was empty.

[image]

gregdavill, 2021-04-11:
I've also removed a `SetToolTip` call, and added a thin shim class 'DialogShim' so we can deal with `SetSizeHintsSz` automatically without manual change of the generated code.

robin7331, 2021-04-11:
The `DialogShim` seems to have an error.

[image]

gregdavill, 2021-04-11:
Sorry, try now.
Also, thanks for helping debug this on OSX. :)

robin7331, 2021-04-11:
Lets make it happen :)
There still are a few tool tips you might have forgotten to remove 🤓
[image]

[image]d

Also I had to modify the `StartOnce` call to the `wx.Timer`.
For some reason it did not like this one.
Thankfully `self.timer.Start(milliseconds=250, oneShot=True)` does the same thing. 

PR submitted 👋

gregdavill, 2021-04-11:
Might need to be careful with `Start` instead of `StartOnce`. But useful to know that we need to avoid 'StartOnce'.

closed 2021-04-11 by gregdavill

gregdavill, 2021-04-11:
Whoops, I'm blind. 
I see you're also got `OnShot=True`. So no functional difference.

robin7331, 2021-04-11:
According to [here](https://docs.wxwidgets.org/3.0.4/classwx_timer.html#a8df981f7075786811598828af1d294ac) `StartOnce` is only a wrapper for `Start` with the wxTIMER_ONE_SHOT parameter.

PR #28 (2021-04-11): Fix wx macOS incompatibility
Removed the remaining tool tips. 
Also I had to modify the `StartOnce` call to the `wx.Timer`.
For some reason it did not like this one.
Thankfully `self.timer.Start(milliseconds=250, oneShot=True)` does the same thing. 

This fixes #27

I am successfully using KiBuzzard now on macOS with KiCad Nightly 5.99 🚀

[image]