gregdavill/KiBuzzard #30: TypeError in plugin.py after running KiCad's Refresh Plugins

robin7331, 2021-04-11:
When I hit `Refresh Plugins` the KiBuzzard Icon disappears from the toolbar and only reappears when I relaunch KiCad. 

[image]

After refreshing the plugins I am still able to try to launch the plugin from the scripting console but I get a TypeError. 
 
[image]

By the way, this is a nice way to test the code without restarting KiCad or being able to successfully reload the plugins. To run the code simply `import KiBuzzard` and then run the plugin via `KiBuzzard.KiBuzzard.plugin.Run()`

gregdavill, 2021-04-11:
Ahh, neat! Thanks for digging into this.

I do have the same issue, and I see other plugins don't do this, so this is probably something we can fix.

robin7331, 2021-04-12:
I renamed the issue to a more appropriate title. The real problem is that this plugin disappears when refreshing the plugins. 
That it might be because of a `TypeError` is only a guess 😋

I'll look into this again today.

gregdavill, 2021-04-12:
Hmm. This morning I couldn't repo this, the button never disappeared.

But just then I did have it happen just now. But after restarting KiCad It's now not disappearing...

robin7331, 2021-04-12:
That's weird. Happens to me all the time. 
On what python version is this stuff running on your end?

gregdavill, 2021-04-12:
Yeah, I've seen it happen to me too. Weird that it's now in a state that works....

I'm on `Python 3.8.6 (default, Jan 27 2021, 15:42:20)`
KiCad-Nightly: `Version: 5.99.0-unknown-675444a646~125~ubuntu20.10.1, release build`

I was going to try that. But I thought I needed to be in the correct directory for `import KiBuzzard` to work...

robin7331, 2021-04-12:
Opening the scripting console and running `import KiBuzzard` should work regardless of the directory since it was loaded as a module.

gregdavill, 2021-04-14:
Can you try with 6218cf3d37e03e77f9da3dfba6e451cbdc8abbea?

I might need to deep dive into how the python plugins are imported. 
But I've tweaked the initial import slightly, seems to work here. But maybe I'm just not hitting the issue...

robin7331, 2021-04-14:
Wohooo! It seems fixed! 🚀
I can refresh the plugins and it still exists. 🤩

And the best part is: It also takes code changes into consideration. No more having to restart KiCad when some python code has changed. Extremely good for the dev workflow.

Great job Greg!

closed 2021-04-14 by gregdavill