gregdavill/KiBuzzard #31: Preview error on Windows

gregdavill, 2021-04-12:
Default window shows no preview of text
[image]

However if you scale the same window you can start to see the rendered text.
Not sure if this is a layering issue, or axis/alignment issue.
[image]

CaptClaude, 2021-04-14:
Datapoint: MacOS 10.14.6 / KiCad Version: (5.1.6-0-10_14), release build does not exhibit this behavior. Freshly installed on 4/13/2021.
[image]

gregdavill, 2021-04-14:
Nice thanks! 

I suspect it's a quirk on windows.

Unrelated note the font kerning could use a bit of work. the `.` in `1.5` isn't very centered.

CaptClaude, 2021-04-14:
Also, notice that the dash ("-") is shown as if it were an underscore ("_"). If you want, I can post what all the various fonts look like on my system. Not all of them are useful, but this is a gigantic step forward from the last version I tried. Anything I can do to help, because this is super useful.

gregdavill, 2021-04-14:
Yep, I did notice that. I've just opened #37 to track that issue.

I think I see the issue in the way we're rendering characters from freetype.

gregdavill, 2021-04-14:
Fixed with c3ba570 

[image]

closed 2021-04-14 by gregdavill