gregdavill/KiBuzzard #32: Settings are not stored or correctly loaded when reopening the dialog

robin7331, 2021-04-12:
It would be nice if the settings would be stored to disk when closing the dialog with `Ok`. 
Right now I have to reenter every value when I like to do many similar labels. 

I see `loadConfig()` and `saveConfig()` I the `dialog.py` file. 
I may misinterpret their intended behavior but if those are supposed to store/load those settings it may be a bug.
Because that is not working on my machine. 

@gregdavill is that a bug or is there nothing like that implemented?

gregdavill, 2021-04-12:
You're right, there are those functions.

They should work as you describe, but they're not 100% connected and working yet.

robin7331, 2021-04-12:
Alright. Just wanted to make sure.
So this is also not working properly on your side? Or is it a macOS thing again?
Might look into it today and submit a PR if I get it to work.

gregdavill, 2021-04-12:
Hopefully this commit will fix this :)

closed 2021-04-12 by gregdavill

robin7331, 2021-04-12:
Indeed! ✌️

commit 5ce0fb0f (2021-04-12): Fix Settings save/restore

Fixes #32