gregdavill/KiBuzzard #35: Plugin fails to load on Windows Nightly

0xdec, 2021-04-13:
Plugin fails to load, both on startup and after refreshing plugin list.

### KiCad Version Info
```
Application: KiCad PCB Editor

Version: (5.99.0-10265-g4c914421a1), release build

Libraries:
	wxWidgets 3.1.4
	libcurl/7.74.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19042), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Apr 12 2021 12:05:27
	wxWidgets: 3.1.4 (wchar_t,STL containers)
	Boost: 1.75.0
	OCC: 7.5.0
	Curl: 7.74.0-DEV
	ngspice: 34
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

### Traceback
```
  File "C:\Program Files\KiCad\5.99\bin\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:/Users/jordi/Documents/KiCad/5.99/scripting\plugins\KiBuzzard\__init__.py", line 1, in <module>
    from . import KiBuzzard
  File "C:/Users/jordi/Documents/KiCad/5.99/scripting\plugins\KiBuzzard\KiBuzzard\__init__.py", line 59, in <module>
    if not plugin.pcbnew_icon_support and not sys.platform.startswith('linux'):
NameError: name 'plugin' is not defined
```

robin7331, 2021-04-13:
Do you have a `error.log` file in your KiBuzzard plugin directory under `{kicad plugins}\KiBuzzard\KiBuzzard\error.log` ?
If yes please post the contents.

robin7331, 2021-04-13:
I am not a windows expert, but why is your path containing forward **_and_** backward slashes?
`"C:/Users/jordi/Documents/KiCad/5.99/scripting\plugins\KiBuzzard\KiBuzzard\__init__.py"`

0xdec, 2021-04-13:
> 
> 
> Do you have a `error.log` file in your KiBuzzard plugin directory under `{kicad plugins}\KiBuzzard\KiBuzzard\error.log` ?
> If yes please post the contents.

### error.log:
```
IndentationError('unexpected unindent', ('C:/Users/jordi/Documents/KiCad/5.99/scripting\\plugins\\KiBuzzard\\KiBuzzard\\plugin.py', 106, 4, '    def InitLogger(self):\n'))
```

0xdec, 2021-04-13:
> 
> 
> I am not a windows expert and this might be unrelated but why is your path containing forward **_and_** backward slashes?
> `"C:/Users/jordi/Documents/KiCad/5.99/scripting\plugins\KiBuzzard\KiBuzzard\__init__.py"`

That's a good question, I have no idea... all of my KiCad Environment Variables use backslashes, so idk where those forward slashes are coming from...

0xdec, 2021-04-13:
Might be unrelated, but I think the following line should probably use `os.path.join()`

https://github.com/gregdavill/KiBuzzard/blob/079327a3c2f152fef4566aee0fe07b90e72de856/KiBuzzard/__init__.py#L27

robin7331, 2021-04-13:
I think I fixed the issue.
Could you please try pulling KiBuzzard from this PR: https://github.com/gregdavill/KiBuzzard/pull/36 and confirm that it is fixed? 😋

robin7331, 2021-04-13:
> Might be unrelated, but I think the following line should probably use os.path.join()

Yeah that's probably a good idea 😅

0xdec, 2021-04-13:
> 
> 
> I think I fixed the issue.
> Could you please try pulling KiBuzzard from this PR: #36 and confirm that it is fixed? 😋

Works for me, thanks!

gregdavill, 2021-04-13:
Whoops. Apologizes. 
I did hit this issue too, and fixed it but didn't push the fix yesterday -_-*

gregdavill, 2021-04-13:
> Might be unrelated, but I think the following line should probably use `os.path.join()`
> 
> https://github.com/gregdavill/KiBuzzard/blob/079327a3c2f152fef4566aee0fe07b90e72de856/KiBuzzard/__init__.py#L27

Thanks! I've changed this too 7506e47

closed 2021-04-13 by gregdavill