gregdavill/KiBuzzard #37: Font kerning is handled incorrectly

gregdavill, 2021-04-14:
Here is an example of input text `AaBb_- [efg]`

[image]

When rendered with GIMP, it looks like this:
[image]

You can see there are various glyph placement differences.

CaptClaude, 2021-04-14:
So my "dash rendered as underscore" issue on MacOS is clearly not a **_Windows_** quirk.

gregdavill, 2021-04-14:
Sorry, I was referring to the preview appearing blank, as a windows quirk. (Appears to be painting below a 'wx.frame')

The kerning issues are a real bug and not OS dependent.

gregdavill, 2021-04-19:
Fixed by #39

closed 2021-04-19 by gregdavill