gregdavill/KiBuzzard #38: Parse string error when relaunching KiBuzzard with an empty text field

robin7331, 2021-04-14:
When you have a fresh start of KiCad, you open KiBuzzard, leave everything as it is and close it, it will have this error when you reopen KiBuzzard again. 

I assume it stores the empty text into the settings and then has a string parsing error when restarting and loading the settings. This is a little tricky to reproduce but occurs sometimes. I think a check wether the string is empty before generating the preview might fix it. 

[image]

**Traceback**
```
Traceback (most recent call last):
File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 122, in ReGeneratePreview
self.polys = self.buzzard.generate(self.m_MultiLineText.GetValue())
File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/buzzard/buzzard.py", line 52, in generate
paths, attributes, svg_attributes = string2paths(self.renderLabel(inString).tostring())
File "/Users/robin/Documents/KiCad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/buzzard/modules/svgstring2path.py", line 110, in string2paths
doc = parseString(svg_string)
File "/private/var/folders/0_/50zzr6qs1v90gkw4sg2qkfph0000gn/T/AppTranslocation/DD653CE0-1920-4826-9541-842706E1A454/d/KiCad.app/Contents/Frameworks/Python.framework/Versions/2.7/lib/python2.7/xml/dom/minidom.py", line 1927, in parseString
from xml.dom import expatbuilder
ImportError: cannot import name expatbuilder
```

gregdavill, 2021-04-14:
Yep, I think your diagnosis is correct. 
When `m_MultiLineText` and `ReGeneratePreview` is called it produces this error.

robin7331, 2021-04-14:
This might also be a good situation for including a [Dialog](https://docs.wxwidgets.org/3.0.4/classwx_dialog.html) if the entered text is longer than x chars to prevent any crazy preview rendering from happening when you accidentally paste long test as you mentioned in #34 ?

gregdavill, 2021-04-14:
Good idea! Added in 6a38106007782ce2781be51fe3b96dc81964e260 and 98883e052a326fb46cd69417a20663c3c16ea51f

robin7331, 2021-04-14:
Sweet. The issue seems fixed. Cannot reproduce the problem anymore.

gregdavill, 2021-04-14:
Cool! :+1:

closed 2021-04-14 by gregdavill