gregdavill/KiBuzzard #41: Plugin not showing up

gregdavill, 2021-04-19:
Hmmm... Did that and the plugin does not show up after closing/opening KiCAD and Refresh Plugins. Delete everything just cloned and restore what I moved out (just in case) and the plugin is back.  

I will take another look at this again tomorrow.

_Originally posted by @CaptClaude in https://github.com/gregdavill/KiBuzzard/issues/39#issuecomment-822156541_

gregdavill, 2021-04-19:
Just moved this over to a new issue.

This _should_ be enough to get this working.
```
greg@greg-xps:/tmp$ git clone https://github.com/gregdavill/KiBuzzard.git --recursive
Cloning into 'KiBuzzard'...
remote: Enumerating objects: 555, done.
remote: Counting objects: 100% (297/297), done.
remote: Compressing objects: 100% (140/140), done.
remote: Total 555 (delta 197), reused 244 (delta 157), pack-reused 258
Receiving objects: 100% (555/555), 3.13 MiB | 2.17 MiB/s, done.
Resolving deltas: 100% (278/278), done.
Submodule 'KiBuzzard/deps/fonttools' (https://github.com/fonttools/fonttools.git) registered for path 'KiBuzzard/deps/fonttools'
Submodule 'KiBuzzard/deps/svg2mod' (https://github.com/gregdavill/svg2mod.git) registered for path 'KiBuzzard/deps/svg2mod'
Cloning into '/tmp/KiBuzzard/KiBuzzard/deps/fonttools'...
remote: Enumerating objects: 58356, done.        
remote: Counting objects: 100% (1954/1954), done.        
remote: Compressing objects: 100% (555/555), done.        
remote: Total 58356 (delta 1318), reused 1860 (delta 1285), pack-reused 56402        
Receiving objects: 100% (58356/58356), 19.99 MiB | 1.39 MiB/s, done.
Resolving deltas: 100% (37269/37269), done.
Cloning into '/tmp/KiBuzzard/KiBuzzard/deps/svg2mod'...
remote: Enumerating objects: 783, done.        
remote: Counting objects: 100% (303/303), done.        
remote: Compressing objects: 100% (178/178), done.        
remote: Total 783 (delta 189), reused 219 (delta 124), pack-reused 480        
Receiving objects: 100% (783/783), 280.84 KiB | 890.00 KiB/s, done.
Resolving deltas: 100% (478/478), done.
Submodule path 'KiBuzzard/deps/fonttools': checked out 'bcba6abce5c0ee16e8083ba669314de4f947247d'
Submodule path 'KiBuzzard/deps/svg2mod': checked out '8a59e7682988b7031d151ad168b109589bfe561c'
```

As you can see it pulls in the two submodules.

If it's not showing an icon, can you open up the pcbnew scripting console and type
```python
import pcbnew
print(pcbnew.FULL_BACK_TRACE)
```
This should print out if there was an error loading the plugin

CaptClaude, 2021-04-20:
Removed working plugin by moving to another location. Proceeded as before:
```
Bardolino:plugins jm$ mkdir KiBuzzard
Bardolino:plugins jm$ cd KiBuzzard
Bardolino:KiBuzzard jm$ git clone https://github.com/gregdavill/KiBuzzard.git --recursive
Cloning into 'KiBuzzard'...
remote: Enumerating objects: 572, done.
remote: Counting objects: 100% (314/314), done.
remote: Compressing objects: 100% (150/150), done.
remote: Total 572 (delta 211), reused 254 (delta 164), pack-reused 258
Receiving objects: 100% (572/572), 3.13 MiB | 2.59 MiB/s, done.
Resolving deltas: 100% (292/292), done.
Submodule 'KiBuzzard/deps/fonttools' (https://github.com/gregdavill/fonttools.git) registered for path 'KiBuzzard/deps/fonttools'
Submodule 'KiBuzzard/deps/svg2mod' (https://github.com/gregdavill/svg2mod.git) registered for path 'KiBuzzard/deps/svg2mod'
Cloning into '/Users/jm/Library/Preferences/kicad/scripting/plugins/KiBuzzard/KiBuzzard/KiBuzzard/deps/fonttools'...
remote: Enumerating objects: 58368, done.        
remote: Counting objects: 100% (1966/1966), done.        
remote: Compressing objects: 100% (567/567), done.        
remote: Total 58368 (delta 1318), reused 1861 (delta 1284), pack-reused 56402        
Receiving objects: 100% (58368/58368), 20.01 MiB | 2.60 MiB/s, done.
Resolving deltas: 100% (37269/37269), done.
Cloning into '/Users/jm/Library/Preferences/kicad/scripting/plugins/KiBuzzard/KiBuzzard/KiBuzzard/deps/svg2mod'...
remote: Enumerating objects: 783, done.        
remote: Counting objects: 100% (303/303), done.        
remote: Compressing objects: 100% (178/178), done.        
remote: Total 783 (delta 189), reused 219 (delta 124), pack-reused 480        
Receiving objects: 100% (783/783), 280.84 KiB | 1.82 MiB/s, done.
Resolving deltas: 100% (478/478), done.
Submodule path 'KiBuzzard/deps/fonttools': checked out 'c6699d10030a7569dc83cb868be9d01a77e8ad0f'
Submodule path 'KiBuzzard/deps/svg2mod': checked out '8a59e7682988b7031d151ad168b109589bfe561c'
Bardolino:KiBuzzard jm$ 
```
No surprises. Open KiCAD, open pcbnew, refresh plugins, no KiBuzzard.
Open scripting console, copy/paste:
```
Py 0.9.8
Python 2.7.16 (default, May 14 2020, 09:34:58) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /Users/jm/Library/Preferences/kicad/PyShell_pcbnew_startup.py
>>> import pcbnew
>>> print(pcbnew.FULL_BACK_TRACE)

>>>
```
Not sure where to look for errors, but this is in the namespace window:
```
locals()

Type: <type 'dict'>

Value: {'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fc3be060200> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fc3bfb62200> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0x7fc3c2960f40> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0x7fc3c295d7f0> >, 'pcbnew': <module 'pcbnew' from '/private/var/folders/0n/gpyffjwd0pvbnf6jpbz9j9bc0000gp/T/AppTranslocation/3DB4D145-EB40-472C-97C2-63BCFD34925C/d/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.pyc'>, '__name__': '__main__', '__doc__': None}
```
Dispatcher window:
```
'EditorChange' from <wx.py.editor.EditorNotebook; proxy of <Swig Object of type 'wxNotebook *' at 0x7fc3c2985aa0> >
'Interpreter.push' from <wx.py.interpreter.Interpreter instance at 0x1136ecbd8>
'Shell.addHistory' from Anonymous
'Interpreter.push' from <wx.py.interpreter.Interpreter instance at 0x1136ecbd8>
'Shell.addHistory' from Anonymous
```
I do not feel much enlightened by this exercise...  Suggestions?

gregdavill, 2021-04-20:
Hmm. Yea. that looks like a dead-end.
In the scripting window can run the following:
```python
import KiBuzzard
KiBuzzard.plugin.Run()
```

On my system this runs the plugin, so maybe it will help us track down the issue.

CaptClaude, 2021-04-20:
More debugging needed on my side.
```
>>> import KiBuzzard
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ImportError: No module named KiBuzzard
>>> 
```
This is the hierarchy of the plugin.  
[image]
If I remove the KiBuzzard folder from the scripting folder and replace it with the folder that was there previously (that worked) it works.
This is the working folder structure.  There are differences. Are those differences user error? Caused by incorrectly cloning the repo? I create a folder called KiBuzzard in the plugins folder. Then I change into that folder and execute the `git` command. 
[image]

So I started again, creating KiBuzzard, cd KiBuzzard, git clone... and when done, I renamed the KiBuzzard created and moved all the files in it up a level and it works.  So I am apparently unable to follow directions or the git clone process is not doing what I/you expect.

One more time, deleted KiBuzzard tree and while in plugins, ran the git clone.  That worked.  Must have been my failure to understand.

gregdavill, 2021-04-20:
Ahh, okay.

So you were installing the plugin 1 extra folder deep? That's an easy mistake to make.

I'll add some extra words to the README to help make it clearer exactly where the plugin should be installed.

closed 2021-04-20 by gregdavill