gregdavill/KiBuzzard #42: Error with python 3.9

gregdavill, 2021-04-19:
A user reports via email that this plugin is not working on Debian 11.

A included log file has this error:
```
04-19 11:38:45 STDERR 138:Traceback (most recent call last):
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/ttFont.py", line 373, in __getitem__
04-19 11:38:45 STDERR 138:    return self.tables[tag]
04-19 11:38:45 STDERR 138:KeyError: 'glyf'
04-19 11:38:45 STDERR 138:
04-19 11:38:45 STDERR 138:During handling of the above exception, another exception occurred:
04-19 11:38:45 STDERR 138:Traceback (most recent call last):
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/ttFont.py", line 373, in __getitem__
04-19 11:38:45 STDERR 138:    return self.tables[tag]
04-19 11:38:45 STDERR 138:KeyError: 'loca'
04-19 11:38:45 STDERR 138:
04-19 11:38:45 STDERR 138:During handling of the above exception, another exception occurred:
04-19 11:38:45 STDERR 138:Traceback (most recent call last):
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 136, in ReGeneratePreview
04-19 11:38:45 STDERR 138:    self.polys = self.buzzard.generate(self.m_MultiLineText.GetValue())
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/buzzard.py", line 47, in generate
04-19 11:38:45 STDERR 138:    self.svgText = self.renderLabel(inString)
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/buzzard.py", line 73, in renderLabel
04-19 11:38:45 STDERR 138:    t.convert_to_path()
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/svg2mod/svg2mod/svg/svg.py", line 1249, in convert_to_path
04-19 11:38:45 STDERR 138:    try: glf = ttf.getGlyphSet()[ttf.getBestCmap()[ord(char)]]
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/ttFont.py", line 678, in getGlyphSet
04-19 11:38:45 STDERR 138:    glyphs = _TTGlyphSet(self, self["glyf"], _TTGlyphGlyf)
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/ttFont.py", line 392, in __getitem__
04-19 11:38:45 STDERR 138:    table.decompile(data, self)
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/tables/_g_l_y_f.py", line 57, in decompile
04-19 11:38:45 STDERR 138:    loca = ttFont['loca']
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/ttFont.py", line 392, in __getitem__
04-19 11:38:45 STDERR 138:    table.decompile(data, self)
04-19 11:38:45 STDERR 138:  File "/home/iz2rpn/.kicad_plugins/KiBuzzard/KiBuzzard/buzzard/../deps/fonttools/Lib/fontTools/ttLib/tables/_l_o_c_a.py", line 23, in decompile
04-19 11:38:45 STDERR 138:    locations.fromstring(data)
04-19 11:38:45 STDERR 138:AttributeError: 'array.array' object has no attribute 'fromstring'
```

Python 3.9 has changed some API and no longer supports this `fromstring`, attribute.

dcelectr, 2021-04-20:
We have been having issues with Python 3.9 on Windows10 with the VedantParanjape fork. 
So, I have advised several people to downgrade their Python to version 3.8, and that has resolved their issue.

gregdavill, 2021-04-20:
I've installed Debian 11, and managed to hit another two instances of this error. With those patched up it's working as expected!

[image]

closed 2021-04-20 by gregdavill