gregdavill/KiBuzzard #45: Alignment not working?

adamgarbo, 2021-05-06:
Hi @gregdavill,

Many thanks for this great plugin!

Working in KiCad (Pcbnew) v5.1.9-0-10_14 and one issue I've noticed is that regardless of the selected alignment option, the labels are always centre-aligned. I'm curious if you've experienced the same problem, or if perhaps this isn't an issue with Nightly.

Cheers,
Adam

gregdavill, 2021-05-06:
I've tried on KiCad 5.1 here, and it's working as expected. 
To use the alignment feature you do need to set a width.

Maybe there is an issue on the MacOS side? Can you see any errors in the kibuzzard.log file that's created inside the plugin folder after you try to use the alignment options?

Note the origin point of the footprint is always the center, use the builtin KiCad Alignment tools to align a group a of labels
[image]

[image]

adamgarbo, 2021-05-06:
Hi Greg,

Thanks for your reply. I do see now that the alignment is working as expected for single lines of text when the width is set sufficiently long.

Should alignment work when entering multiple lines? It seems as if the alignment setting may only be applied to the last line:

[image]

Cheers,
Adam

gregdavill, 2021-05-06:
Ahh, I see now.

Right now all the text is generated and then alignment is performed, so it's probably aligning based on the longest line.
A quick fix is to generate 3 separate labels and just arrange them like this on the PCB.

gregdavill, 2022-06-18:
The recommended workaround is to generate separate labels and use KiCads own groups/alignment tools to position them.

Adding extra text alignment features isn't on my roadmap for this tool. However I would accept pull requests if someone else wanted to implement these features.

closed 2022-06-18 by gregdavill