gregdavill/KiBuzzard #46: 5.99: "Not in schematic" attribute should probably be set

alexvlockwood-bes, 2021-05-07:
Since the labels are pasted directly into the layout, they don't correspond to anything in the schematic, and thus usually get deleted on the next sch→pcb sync. If "Not in schematic" is set, they are kept.

gregdavill, 2021-05-08:
I agree! This is only an option in 5.99 though right? Along with `exclude from BOM` which probably also makes sense for these logos.

A quick work-around is to make sure `Delete footprints with no symbols` is not checked when updating the PCB.
[image]

gregdavill, 2021-05-08:
It looks like we're originally using `(attr virtual)` which was being extended to the newer v6 syntax of `(attr exclude_from_pos_files exclude_from_bom)`

On v6 the labels should new be using: `(attr board_only exclude_from_pos_files exclude_from_bom)`

closed 2021-05-13 by gregdavill