gregdavill/KiBuzzard #47: 5.99: doesn't work until Python console has been opened

alexvlockwood-bes, 2021-05-07:
On my 5.99 setup, KiBuzzard fails to generate the footprint (OK button just does nothing) unless the Python console has been opened at least once in that session.

If I can provide any more relevant info, let me know! I don't see any error messages show up anywhere.

```
Application: KiCad PCB Editor

Version: 5.99.0-unknown-10519+g68085820ff+buster, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3

Platform: Linux 5.9.0-0.bpo.5-amd64 x86_64, 64 bit, Little endian, wxGTK, , x11

Build Info:
	Date: May  6 2021 11:22:35
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.67.0
	OCC: 7.3.0
	Curl: 7.64.0
	ngspice: 30
	Compiler: GCC 8.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

gregdavill, 2021-05-08:
Interesting I've not seen this behavior before on my test setups.

When KiBuzzard runs it generates a `kibuzzard.log` file with some debug info. Is there anything interesting in there when you attempt to make a footprint without the console open?

One thought. The plugin uses a a wx_widgets call to switch focus to pcbnew and run a `Ctrl+V` to enable interactive placement of a label, maybe this isn't working correctly with the window manager you're using on Debian?

alexisvl, 2021-05-08:
oh no i couldn't resist hopping on my personal account at midnight on a friday night to follow up lol

WM is just xfce, looks like an environment variable isn't getting set so it's not showing up in the version info.

But I have a fix! Sort of. I doubt this is the _proper_ fix since what you have looks like it's supposed to work according to the docs, but hopefully, it's a pointer in the right direction. I don't really know wxpython. If I make this change, it works

```
diff --git a/KiBuzzard/plugin.py b/KiBuzzard/plugin.py
index b48a1d4..b0bc337 100644
--- a/KiBuzzard/plugin.py
+++ b/KiBuzzard/plugin.py
@@ -83,9 +83,9 @@ class KiBuzzardPlugin(pcbnew.ActionPlugin, object):
                 footprint_string = p_buzzard.create_v6_footprint()
 
 
-                if wx.TheClipboard.Open():
-                    wx.TheClipboard.SetData(wx.TextDataObject(footprint_string))
-                    wx.TheClipboard.Close()
+                if wx.Clipboard.Get().Open():
+                    wx.Clipboard.Get().SetData(wx.TextDataObject(footprint_string))
+                    wx.Clipboard.Get().Close()
                     
             dlg.EndModal(wx.ID_OK)
 
```

alexisvl, 2021-05-08:
(btw this is what's in `kibuzzard.log`. Thanks for pointing out that file

```
05-08 00:32:01 STDERR 138:Traceback (most recent call last):
05-08 00:32:01 STDERR 138:  File "/home/alex/.local/share/kicad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/dialog/dialog.py", line 255, in OnOkClick
05-08 00:32:01 STDERR 138:self.func(self, self.buzzard)
05-08 00:32:01 STDERR 138:  File "/home/alex/.local/share/kicad/5.99/scripting/plugins/KiBuzzard/KiBuzzard/plugin.py", line 86, in run_buzzard
05-08 00:32:01 STDERR 138:if wx.TheClipboard.Open():
05-08 00:32:01 STDERR 138:  File "/usr/lib/python3/dist-packages/wx/core.py", line 1337, in __getattr__
05-08 00:32:01 STDERR 138:return getattr(self._instance, name)
05-08 00:32:01 STDERR 138:AttributeError
05-08 00:32:01 STDERR 138::
05-08 00:32:01 STDERR 138:'NoneType' object has no attribute 'Open'
```

closed 2021-05-13 by gregdavill