gregdavill/KiBuzzard #49: Footprint REF & VAL, alignment, and a possible bug when changing height

elekgeek, 2021-05-25:
Thank you for the great work done on this plugin, really appreciated.

Is it possible to have an option not to generate the reference and value for the footprint generated, it would be great, as it is visually messy and not really required after all.

Second thing, alignment for multiple lines doesn't work.

Thirdly, is that when changing height, no footprint is generated, I have to close the plugin and re-poen it to generate a footprint.

thanks!

gregdavill, 2021-05-26:
> Thank you for the great work done on this plugin, really appreciated.

Thanks :)

> Is it possible to have an option NOT to generate the reference and value for the footprint generated, it would be great, as it is visually messy and not really required after all.

Yep we should be able to remove the ref/values. These are hidden, but do still cause a bit of visual clutter.

> Second thing, alignment for multiple lines doesn't work.

Alignment is already a tracked issue here: https://github.com/gregdavill/KiBuzzard/issues/45

> Thirdly, is that when changing height, no footprint is generated, I have to close the plugin and re-poen it to generate a footprint.

Hmm, interesting. Height seems to work for me, When the footprint isn't generated, can you look in the `kibuzzard.log` file that should be created in the plugin folder?

elekgeek, 2021-06-10:
sure, once I'm home next week :)

elekgeek, 2021-06-10:
Thank you for your prompt answer.

> > Is it possible to have an option NOT to generate the reference and value for the footprint generated, it would be great, as it is visually messy and not really required after all.
> 
> Yep we should be able to remove the ref/values. These are hidden, but do still cause a bit of visual clutter.
> 

I assume leaving them there is OK, but with empty values.

> Hmm, interesting. Height seems to work for me, When the footprint isn't generated, can you look in the `kibuzzard.log` file that should be created in the plugin folder?

sure, once I'm home next week :)

elekgeek, 2021-06-13:
Hello Greg,

I start the plugin, change the height, click OK, nothing on the PCB appears.

Here is the log:

```
06-14 00:18:19 root 1331:Loading system fonts.
06-14 00:18:20 root 1349:  Found 183 fonts in system
06-14 00:18:21 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:18:21 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:18:21 root 1251:Unsuported character in <text> element "
"
06-14 00:18:21 unfiltered 871:Found SVG layer: F.SilkS
06-14 00:18:21 root 458:  Inlining 28 segments...
06-14 00:18:26 root 928:  Writing Text with 1195 points
06-14 00:18:26 root 458:  Inlining 1 segments...
06-14 00:18:26 root 928:  Writing Text with 56 points
06-14 00:18:26 root 928:  Writing Text with 21 points
06-14 00:18:26 root 928:  Writing Text with 21 points
06-14 00:18:26 root 928:  Writing Text with 19 points
06-14 00:18:26 root 928:  Writing Text with 18 points
06-14 00:18:26 root 928:  Writing Text with 18 points
06-14 00:18:26 root 928:  Writing Text with 18 points
06-14 00:18:26 root 928:  Writing Text with 8 points
06-14 00:18:26 root 928:  Writing Text with 8 points
06-14 00:18:26 root 928:  Writing Text with 4 points
06-14 00:18:29 STDERR 138:Traceback (most recent call last):
06-14 00:18:29 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 144, in labelEditOnText
06-14 00:18:29 STDERR 138:self.ReGeneratePreview()
06-14 00:18:29 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 161, in ReGeneratePreview
06-14 00:18:29 STDERR 138:DefaultPadding = 7.75 * 4 * 0.25 * (1/scale)
06-14 00:18:29 STDERR 138:ZeroDivisionError
06-14 00:18:29 STDERR 138::
06-14 00:18:29 STDERR 138:float division by zero
06-14 00:19:12 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:19:12 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:19:12 root 1251:Unsuported character in <text> element "
"
06-14 00:19:12 unfiltered 871:Found SVG layer: F.SilkS
06-14 00:19:12 root 458:  Inlining 28 segments...
06-14 00:19:17 root 928:  Writing Text with 1195 points
06-14 00:19:17 root 458:  Inlining 1 segments...
06-14 00:19:17 root 928:  Writing Text with 56 points
06-14 00:19:17 root 928:  Writing Text with 21 points
06-14 00:19:17 root 928:  Writing Text with 21 points
06-14 00:19:17 root 928:  Writing Text with 19 points
06-14 00:19:17 root 928:  Writing Text with 18 points
06-14 00:19:17 root 928:  Writing Text with 18 points
06-14 00:19:17 root 928:  Writing Text with 18 points
06-14 00:19:17 root 928:  Writing Text with 8 points
06-14 00:19:17 root 928:  Writing Text with 8 points
06-14 00:19:17 root 928:  Writing Text with 4 points
06-14 00:19:19 STDERR 138:Traceback (most recent call last):
06-14 00:19:19 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 144, in labelEditOnText
06-14 00:19:19 STDERR 138:self.ReGeneratePreview()
06-14 00:19:19 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 161, in ReGeneratePreview
06-14 00:19:19 STDERR 138:DefaultPadding = 7.75 * 4 * 0.25 * (1/scale)
06-14 00:19:19 STDERR 138:ZeroDivisionError
06-14 00:19:19 STDERR 138::
06-14 00:19:19 STDERR 138:float division by zero
06-14 00:23:02 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:23:02 root 1199:Font "FredokaOne" does not natively support style "Regular" using "Path" instead
06-14 00:23:02 root 1251:Unsuported character in <text> element "
"
06-14 00:23:02 unfiltered 871:Found SVG layer: F.SilkS
06-14 00:23:02 root 458:  Inlining 28 segments...
06-14 00:23:07 root 928:  Writing Text with 1195 points
06-14 00:23:07 root 458:  Inlining 1 segments...
06-14 00:23:07 root 928:  Writing Text with 56 points
06-14 00:23:07 root 928:  Writing Text with 21 points
06-14 00:23:07 root 928:  Writing Text with 21 points
06-14 00:23:07 root 928:  Writing Text with 19 points
06-14 00:23:07 root 928:  Writing Text with 18 points
06-14 00:23:07 root 928:  Writing Text with 18 points
06-14 00:23:07 root 928:  Writing Text with 18 points
06-14 00:23:07 root 928:  Writing Text with 8 points
06-14 00:23:07 root 928:  Writing Text with 8 points
06-14 00:23:07 root 928:  Writing Text with 4 points
06-14 00:23:09 STDERR 138:Traceback (most recent call last):
06-14 00:23:09 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 144, in labelEditOnText
06-14 00:23:09 STDERR 138:self.ReGeneratePreview()
06-14 00:23:09 STDERR 138:  File "C:\Users\elekgeek\AppData\Roaming\kicad\scripting\plugins/KiBuzzard-main/KiBuzzard/dialog/dialog.py", line 161, in ReGeneratePreview
06-14 00:23:09 STDERR 138:DefaultPadding = 7.75 * 4 * 0.25 * (1/scale)
06-14 00:23:09 STDERR 138:ZeroDivisionError
06-14 00:23:09 STDERR 138::
06-14 00:23:09 STDERR 138:float division by zero
```

elekgeek, 2021-07-23:
Hi Greg, the logs have been waiting for a long time :D

gregdavill, 2021-07-24:
Can you try these changes? They should fix the height and Ref/Val clutter.

I'm still not sure about how we handle multiple line alignment.

elekgeek, 2021-07-24:
Done, both division by zero and REF/VAL are also solved, thanks!

closed 2021-07-24 by elekgeek