gregdavill/KiBuzzard #50: KiBuzzard Plugin Not showing in KiCad Tools after Installation

dhairyashah1, 2021-06-10:
1. clone the repo in ~/.kicad/scripting/plugins
2. After multiple attempts still the plugin is not showing in KiCAD tools even after refreshing.
3. What could be a possible fix? I'm using Ubuntu 20.04 LTS.
4. Also got this error as well
[image]

gregdavill, 2021-06-10:
Is there a `KiBuzzard.log` or `error.log` file created?

dhairyashah1, 2021-06-10:
@gregdavill  yes there is a kibuzzard.log file at this path ~/.kicad/scripting/plugins/KiBuzzard. But it seems to be empty.

gregdavill, 2021-06-10:
Which version of KiCad do you have installed?

Can you see anything in the `Action Plugin` tab in the KiCad preferences? There should be an icon that appears, but on this page there is an option to not show the icon.
[image]

gregdavill, 2021-06-10:
I've also just noticed that I've got mine installed in `~/.kicad/scripting/`. Both should work, but maybe try that too?

dhairyashah1, 2021-06-11:
> Which version of KiCad do you have installed?
> 
> Can you see anything in the `Action Plugin` tab in the KiCad preferences? There should be an icon that appears, but on this page there is an option to not show the icon.
> [image]

Nope Not getting this. Kicad version 5.1.5. Tried in scripting dir too , but same result.

x70b1, 2021-09-25:
I can see the same issue. After a clean install the plugin is not loading.

Sometimes I have this in my error.log. It means permission denied. 
```
PermissionError(13, 'Keine Berechtigung')
```

`chmod 777 -R /usr/share/kicad/scripting/plugins/KiBuzzard/` is evil, but lets the plugin load at a kicad startup. 
So we need to fix the permission for a small set of files I guess.

gregdavill, 2021-09-26:
You shouldn't see any permission issues if you're installing it into `~/.kicad/Scripting/`, since that's in your home directory.

The only thing we are trying to write to that file is the error logs, and a config file which stores all the current GUI values. Even if we're not able to write it shouldn't cause the plugin to crash and not load, so this is a bug.

x70b1, 2021-09-26:
I install it to `/usr/share/kicad/scripting/plugins/` as it works for every user on the system. Especially in the case of a prepackaged installation this is the way to go.

[InteractiveHtmlBom](https://github.com/openscopeproject/InteractiveHtmlBom) is also there and just works fine. So it would be cool if we can find out what we need to change to run it from there.

But yeah, @dhairyashah1 installed it also in his home dir and it is not loading. So i am not sure if thats the same root cause.

gregdavill, 2021-12-31:
Permission errors should be fixed with: 8c11d36b3bdee416121ac418a5248658b47ee843

closed 2021-12-31 by gregdavill

x70b1, 2021-12-31:
Seems to work. Awesome! :sunglasses: