gregdavill/KiBuzzard #54: "D" in "GND" disappears when using Cap right ")"

alextsam, 2021-07-18:
I encountered the following bizarre issue:

When creating a GND label with the following settings, the result is missing the "D" letter. This does not happen if the character is lowercase (or any other character) or if the cap right is something else than ")".

[image]

[image]

gregdavill, 2021-07-18:
Interesting indeed.
I can re-create this here, looks like the object isn't correctly cutting out the outer `D` shape. But all the paths are there if you turn on outline mode.

[image]

I'll look into why this might be. 

In the mean-time you can manually work-around, by editing the footprint and moving the one of the control points on the bottom left of the `D`.
[image]

alextsam, 2021-07-18:
The manual workaround works well, thanks!

By the way, here is another case I have encountered, in case it helps with researching the root cause. This time, the "H" in "CHG" has the same issue, with left cap "(" and right cap ")":

[image]

gregdavill, 2021-12-27:
Seems like this was fixed at some point... I can't reproduce this anymore
Please open up a new issue if you can repo with newest commit.

closed 2021-12-27 by gregdavill