gregdavill/KiBuzzard #58: Generating labels in 5.99/6.0 doesn't trigger the paste part...

UnexpectedMaker, 2021-11-03:
This is possibly just on macOS, but creating a label does everything but the automatic "paste" part of the process.

Pressing [OK] on the dialog box, just triggers a layer change to a copper layer and nothing else happens, but if you then press `CMD+V`, the label is there and you can place it.

Everything else about it seems to be working.

Tried this with 5.99 nightly from Oct 28/29th `kicad-unified-20211028-175635-b42fce430d.dmg` and from the latest nightly today `kicad-unified-20211102-155217-9e760512ac.dmg` with the same results.

gregdavill, 2021-11-03:
Once a footprint is generated it's placed onto the clipboard. This step sounds like it's all working. As you say manually pasting from the clipboard works.

In order to start an interactive placement in KiCad I'm adding "CTRL + V" to the input keyboard queue.
https://github.com/gregdavill/KiBuzzard/blob/8bf3db0e7950d92f59178c18e60e7bb816d5173f/KiBuzzard/plugin.py#L96-L102

It might be an issue with CMD != CONTROL. But the API reference seems to indicate that they're equivalent.
https://wxpython.org/Phoenix/docs/html/wx.KeyModifier.enumeration.html#wx-keymodifier

But since you say that your layer copper layer is changing. This sounds like your pcbnew window is seeing just the `V` key..
[image]

gregdavill, 2021-11-03:
I see in the unittests for wx that they skip UISimulatorAction.Char on OSX. So maybe there is a bug there?
https://github.com/wxWidgets/Phoenix/blob/master/unittests/test_uiaction.py#L172

I've switched to using KeyUp/KeyDown as a test

gregdavill, 2021-12-27:
Is this still an issue on OSX?

bwiessneth, 2022-01-01:
Installed your plugin today (1.0.0) and seeing the same issue as reported by @UnexpectedMaker.
Pressing CMD + V after the dialog closes will paste the generated footprint.

gregdavill, 2022-01-01:
Are you also on OSX? I suspect either the paste operation doesn't work. Or it's been captured by a different window. :/

gregdavill, 2022-01-01:
We have an old macbook in the office. If it's compatible with Catalina I'll be able to load up KiCad v6 and try doing some debugging myself over the next week.

Maybe in the meantime we can disable it trying to automatically place? and show a message to OSX users at the bottom of the KiBuzzard dialog?
```
Note: On OSX the footprint will be placed in the clipboard. Use `CMD + V` to place it on your board
```

bwiessneth, 2022-01-02:
Yes, running KiCad 6.0 on Catalina.
Displaying the note sounds like a good alternative

gregdavill, 2022-01-04:
Using the latest commit, and `v1.0.0` from PCM.
I'm able to get the automatic paste working on Catalina (The newest OSX that my macbook pro I'm borrowing supports :100: )

The first time I ran the plugin I got a popup telling me that KiCad wanted to use accessibility features. And I needed to enable it to do so. This is likely due to the way we're currently emulating the paste operation.
[image]

On the first time through the app it doesn't work. On the second time it does. So something still wrong here, I'll investigate if we can place key events a bit more directly to the window handler, or otherwise paste our footprint. 

https://user-images.githubusercontent.com/344310/148063350-54f48be3-2501-4fba-aecd-a6fe644bb0bc.mp4

cwrb8888, 2022-01-22:
I'm using `v1.0.0` from PCM, and the same problem cased in `Windows 10 21H2`

[image]

[image]

Thanks for your work.

gregdavill, 2022-02-22:
On Windows/OSX we now bypass the need for UIActionSimulator, and insert a "Ctrl + V" directly into the PCB editors Event queue. 

This should fix some of the issues here.

flavioamieiro, 2022-04-09:
Hi there, I just wanted to say that I found the exact same bug on linux on KiBuzzard's `main` branch (78328e91) and Kicad built from master ([`0bd3341f`](https://github.com/KiCad/kicad-source-mirror/commit/0bd3341f0f007300c3cdf66bc0c616669abb760e)).

I'm using i3 on debian, in case that's important information.

I know this is not a huge issue as there is this work around, but I decided to comment here since looking for 'kibuzzard not generating labels' didn't land me anywhere, and I only found this by directly looking in the issues here and I only saw comments about OS X and Windows.

By the way, thanks a lot for KiBuzzard :)
It makes my PCBs look much better and it's nice to see development is still active (which I know is a lot of work).

gregdavill, 2022-06-18:
I think your problem might be related to using i3. 

I saw similar issues when it was my window manager. When we generate a label we try to close dialog window and bring focus to the main KiCad window. This doesn't play nice with i3.

On Windows/MacOS I've now been able to generate an Internal Event and push it onto the main windows event queue, instead of needing to use the system clipboard. This isn't easily possible on linux because the version of wxWidgets is a bit old.

flavioamieiro, 2022-07-05:
@gregdavill that makes sense for the i3 portion of things. Thanks for letting me know!

ccoenen, 2022-08-04:
I'm on Windows 10 and also have to press <kbd>Ctrl</kbd>+<kbd>V</kbd>. In fact I came here, thinking this was broken. This ticket gave me the idea to try the paste command - and this works.

I'm on
* Windows 10 21H2 (19044.1826)
* KiCad 6.0.6
* KiBuzzard 1.2.0 (00168be634)

If you want me to try or double check anything on windows (up to modifying plugin code) let me know!

gregdavill, 2024-05-06:
These issues might all be fixed now that we can add the paste command directly to the wxwidget command queue. Please re-open an issue is this is an issue on the latest release

closed 2024-05-06 by gregdavill