gregdavill/KiBuzzard #59: Re-edit existing label.....

UnexpectedMaker, 2021-11-03:
As per our discussion on discord @gregdavill, I'm just putting this in here as a "possibly one day, maybe in 2027 or something" to re-visit the ability to either be able to edit an existing label, or at least have some viewable meta-data attached to a label footprint to be able to remember what settings were used when generating that specific label.

Cheers!

gregdavill, 2021-11-03:
It's not worth trying to use this branch directly since it's so old. But you can see I did investigate an "edit" feature.
https://github.com/gregdavill/KiBuzzard/tree/feature-edit

The main issue with fully editing footprints was this: https://gitlab.com/kicad/code/kicad/-/issues/7065 
But prefilling the UI with the same parameters used to generate a selected footprint should be possible.

elekgeek, 2021-12-02:
apparently fix committed 3 weeks ago.

gregdavill, 2022-01-01:
I've updated that branch to match recent changes. Looks like it is all fixed upstream in KiCad! 

Here is how it works so far.
- Ensure only 1 KiBuzzard footprint is selected. Open plugin
- Edit, and close. Footprint is updated
[image]

Note: This will only work on footprint generated on this branch, so you can't open up old design and edit KiBuzzard labels.

Still a few more bugs,issues,testing before it's ready to merge into main.

closed 2022-01-01 by gregdavill