gregdavill/KiBuzzard #61: KiBuzzard changes behavior when migrated from 5.99 to 6.0rc1

hwreverse, 2021-11-16:
Hi Greg,

(PS: I wrote aboute it in the KiCAD forum too: [kibuzzard-changed-behavior-between-5-99-and-6-0-rc1](https://forum.kicad.info/t/kibuzzard-changed-behavior-between-5-99-and-6-0-rc1/32008))

On 5.99 Kicad until today, when I clicked on “OK” in the user GUI to create the SilkScreen tag, it “attached” the tag to my mousepointer, so I had just to place it.

On 6.0 rc1, when I clicked “OK” , the GUI closed as usual, and then nothing. First I thought that KiBuzzard was broken on 6.0rc1, but then I thought of pressing Ctrl+V, and there was the tag !

Maybe it is a bug, maybe it is intended, but that’s the behavior (and workaround for now): **The tag is in the clipboard**, safe and sound.

My **5.99 (Windows nightly) version**: `(5.99.0-13222-g16a34e6adb), release build`
My **6.0rc1 (Windows nightly) version:** `(6.0.0-rc1-1-g4df25edb3d), release build`

**kibuzzard.log** in the KiBuzzard plugin folder shows the same stuff on both versions:

`11-16 15:09:23 root 1331:Loading system fonts.`
`11-16 15:09:23 root 1349:  Found 348 fonts in system`
`11-16 15:09:23 STDERR 138:C:\Program Files\KiCad\5.99\share\kicad\scripting\plugins\KiBuzzard\KiBuzzard\dialog\compat.py:17: wxPyDeprecationWarning: Call to deprecated item. Use SetSizeHints instead.`
`11-16 15:09:23 STDERR 138:  super(wx.Dialog, self).SetSizeHintsSz( a,b)`
`11-16 15:09:24 root 1198:Font "FredokaOne" does not natively support style "Regular" using "Path" instead`
`11-16 15:09:24 unfiltered 871:Found SVG layer: F.SilkS`
`11-16 15:09:24 root 458:  Inlining 3 segments...`
`11-16 15:09:24 root 928:  Writing Text with 125 points`
`11-16 15:09:24 root 928:  Writing Text with 8 points`
`11-16 15:09:24 root 928:  Writing Text with 8 points`
`11-16 15:09:25 unfiltered 871:Found SVG layer: F.SilkS`
`11-16 15:09:25 root 458:  Inlining 3 segments...`
`11-16 15:09:25 root 928:  Writing Text with 125 points`
`11-16 15:09:25 root 928:  Writing Text with 8 points`
`11-16 15:09:25 root 928:  Writing Text with 8 points`

--
Best regards (and thank you 1000 times for your great plugin!)

Alexander aka @hwreverse

gregdavill, 2021-11-18:
Can you try 5aa46f56bfe3ad625bc35c47c14e24696bb1b1bc

I think we just weren't checking the new `6.0` build info for the paste action

hwreverse, 2021-11-18:
Tested with 5aa46f5, now it works as before!

Thank you very much, Greg.

Best regards,

Alexander aka @hwreverse

closed 2021-11-18 by hwreverse

elekgeek, 2021-11-18:
I think these two lines should become like this:
```
import re

            #elif '5.99' in self.kicad_build_version or '6.0' in self.kicad_build_version:
            elif re.match('5\.[0-9]{1,2}\.[0-9]{1,2}', self.kicad_build_version):

            #if '5.99' in self.kicad_build_version or '6.0' in self.kicad_build_version:
            if re.match('6\.[0-9]{1,2}(\.[0-9]{1,2})?', self.kicad_build_version):
```