gregdavill/KiBuzzard #62: Does not show up after install on Windows 10?

gcsalzburg, 2021-12-01:
Hi Greg! I followed the steps to install on Windows 10, but I don't see anything when I "Refresh plugins" in KiCad.
The menu is just blank there.

Here you can see where it is installed. Are there any other steps to get it to appear?
[image]

Thanks,
George

arturo182, 2021-12-01:
It should be inside the `scripting` directory. Also best way to make sure plugins are reloaded is to close KiCad and start it again, and not just eeschema or pcbnew, the main KiCad window.

gcsalzburg, 2021-12-01:
Great - that works perfectly now!
Sorry for the stupid question :)

closed 2021-12-01 by gcsalzburg

arturo182, 2021-12-01:
No problem, enjoy :)