gregdavill/KiBuzzard #63: Not working in 6.0.0?

korken89, 2021-12-27:
Hi!

I just upgraded from KiCAD 5 to 6, and the plugin is no longer showing up, and I'm not sure why.
I installed it via Arch AUR, and it was working fine - but not after the upgrade.
The install is located in `/usr/share/kicad/scripting/plugins/` and I also have the `InteractiveHTMLBom` next to `KiBuzzard` (also installed via AUR) and it's showing up correctly in the PCB editor.

```console
> ls /usr/share/kicad/scripting/plugins/
arc_test.py
bga_wizard.py
circular_pad_array_wizard.py
FootprintWizardBase.py
FPC_wizard.py
__init__.py
InteractiveHtmlBom
KiBuzzard
kicad_qrcode.py
microMatch_connectors.py
mutualcap_button_wizard.py
PadArray.py
qfn_wizard.py
qfp_wizard.py
qrcode_footprint_wizard.py
sdip_wizard.py
touch_slider_wizard.py
uss39_barcode.py
zip_wizard.py
```

Any idea on how to find out what the issue is?

I'm running Arch Linux, with KiCAD 6.0.0.

gregdavill, 2021-12-27:
Plugin still loading fine for me on 6.0. (Looks like some changes needed to get it working again on 6.99.)

Is `/usr/share/kicad/scripting/plugins/` is a valid plugin directory? This can be checked using the command in the README.

Using the `kicad-nightly` package in ubuntu I get the following directories
```
>>> import pcbnew; print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
/usr/share/kicad-nightly/scripting
/usr/share/kicad-nightly/scripting/plugins
/home/greg/.config/kicad/6.99/scripting
/home/greg/.config/kicad/6.99/scripting/plugins
/home/greg/.local/share/kicad/6.99/scripting
/home/greg/.local/share/kicad/6.99/scripting/plugins
/home/greg/.local/share/kicad/6.99/3rdparty/plugins
```

I generally install plugins to my home directory.
When you open up preferences what does it look like, if there are errors importing a plugin there will be a warning icon near the bottom of this page.
[image]

gregdavill, 2021-12-27:
Also if you'd installed the plugin into a directory where you didn't have write permissions then the plugin would fail to load. But this is now fixed with the most recent commits.

korken89, 2021-12-28:
When I check `import pcbnew; print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)` I get an empty string:

```console
❯ python
Python 3.10.1 (main, Dec 18 2021, 23:53:45) [GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pcbnew; print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)

```

The preferences page you show is correctly showing the `InteractiveHtmlBom` at the `/usr/share/kicad/scripting/plugins/` path, so I think it's valid:
[image]

It's very weird, because all was working fine in v5, only after the update to v6 did this plugin disappear.

korken89, 2021-12-28:
Ah wait, if I run KiCAD in a terminal I get the following warnings from `KiBuzzard`:

```console
/usr/share/kicad/scripting/plugins/KiBuzzard/buzzard/../deps/svg2mod/svg2mod/svg2mod.py:644: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  elif item.name is not "":
/usr/share/kicad/scripting/plugins/KiBuzzard/buzzard/../deps/svg2mod/svg2mod/svg2mod.py:870: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if name == item.name and item.name is not "":
```

gregdavill, 2021-12-28:
Those warnings aren't an issue.

What does the error window say when you open it up. This button here:
[image]

korken89, 2021-12-28:
Ahh there is the error log!

It says:

```console
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/plugin.py", line 22, in __init__
    self.InitLogger()
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/plugin.py", line 119, in InitLogger
    handler2 = logging.FileHandler(log_file)
  File "/usr/lib/python3.10/logging/__init__.py", line 1167, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.10/logging/__init__.py", line 1197, in _open
    return open_func(self.baseFilename, self.mode,
PermissionError: [Errno 13] Permission denied: '/usr/share/kicad/scripting/plugins/kibuzzard.log'

During handling of the above exception, another exception occurred:

**********************************
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/__init__.py", line 53, in <module>
    with open(log_file, 'w') as f:
PermissionError: [Errno 13] Permission denied: '/usr/share/kicad/scripting/plugins/KiBuzzard/error.log'

**********************************
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/plugin.py", line 22, in __init__
    self.InitLogger()
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/plugin.py", line 119, in InitLogger
    handler2 = logging.FileHandler(log_file)
  File "/usr/lib/python3.10/logging/__init__.py", line 1167, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.10/logging/__init__.py", line 1197, in _open
    return open_func(self.baseFilename, self.mode,
PermissionError: [Errno 13] Permission denied: '/usr/share/kicad/scripting/plugins/kibuzzard.log'

During handling of the above exception, another exception occurred:

**********************************
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/share/kicad/scripting/plugins/./KiBuzzard/__init__.py", line 53, in <module>
    with open(log_file, 'w') as f:
PermissionError: [Errno 13] Permission denied: '/usr/share/kicad/scripting/plugins/KiBuzzard/error.log'
```

gregdavill, 2021-12-28:
Are you using the latest commit? https://github.com/gregdavill/KiBuzzard/commit/c65c93f7e3a819ff02d4b3b425acd6dbf4011b90

I removed the creation of error.log, and tried to fix up the permission issue. But I might have missed something.

korken89, 2021-12-28:
Aha, I needed a reinstall of the plugin! Thanks for the help.
I did not notice the commit was only 11h old, and I reinstalled yesterday ~16h ago - so the commit was missing.

Now it's working fine, only the icon is not working - but that I can live with. :)
[image]

gregdavill, 2021-12-28:
Hmm, the icon _should_  be working.

korken89, 2021-12-28:
I checked the `./kicad` log and there are no errors, and there is no yellow triangle like last time.
Sorry, I'm very unfamiliar with KiCAD debugging so I'm not sure where to look.

closed 2021-12-31 by gregdavill