gregdavill/KiBuzzard #67: Couple of bugs

anishkgt, 2022-01-07:
Once deleting the graphic does not delete it itself but remains. Had to restart kicad to see it work. Moreover once this plugin is open, when closing PCB Editor closes KiCad completely. The next line space is way too much when adding to lines. Otherwise the plugin is helpful

gregdavill, 2022-01-07:
Can you let me know what Version of KiCad you are on? I'd like to be able to re-create these bugs.
The window should be modal, and atleast on Ubuntu I'm not able to close PCB Editor when the KiBuzzard window remains open.

The text-formating options for multiline labels is not great right now. I suggest that creating two seperate labels, and grouping them is probably the better option.

anishkgt, 2022-01-07:
I am using KiCad v6

Regards,George

anishkgt, 2022-01-08:
Here is a [screenshot ](https://ibb.co/TkStJH4)of another error i encountered while editing one of them after placing them

AiGreek, 2022-01-11:
> 
I had the same issue just once, but can't reproduce it either.

Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-294-g04c46a8fd2), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan  7 2022 21:00:04
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

anishkgt, 2022-01-11:
I did reproduce it by clicking edit and then cancel img the window that popped up and then opening kibuzzard gave that error. 

Regards,
George

gregdavill, 2022-02-22:
The graphic remaining after delete, I believe was caused by an issue with the label editing, that should be fixed now.

Still need to investigate attempting to close PCBnew when the KiBuzzard window is open.

closed 2022-06-18 by gregdavill