gregdavill/KiBuzzard #68: Couple of suggestions

anishkgt, 2022-01-08:


gregdavill, 2022-01-12:
Multi line formatting needs work, I'm not sure how useful it really is to implement. 
A simple workflow is to produce separate labels for each line, and then format them on the PCB?

Can you draw a quick sketch with how you'd like top/bottom caps to look? I can't picture what they would look like.

anishkgt, 2022-01-13:
Some thing like this 

Regards,
George

gregdavill, 2022-02-22:
Hi @anishkgt I don't see an image here.

Jkillelea, 2022-04-03:
This seems like the thread for suggestions - I'd love it if there were an ability to tweak labels after they're already placed on the board. Otherwise, on every change, I have to recreate the label from scratch. I know this might be a challenge given that each label is created as its own footprint, but if there's a straightforward way, I could try and implement it myself.

gregdavill, 2022-04-03:
> This seems like the thread for suggestions - I'd love it if there were an ability to tweak labels after they're already placed on the board. Otherwise, on every change, I have to recreate the label [...]

Hi @Jkillelea, the feature as you've described is already implemented. 
You can find a demo video here: https://github.com/gregdavill/KiBuzzard/pull/65 

It's available in the KiCad PCM as a "testing release". Create a new label (Needs to have been created with the newest KiBuzzard). Once placed, with it selected open up KiBuzzard, it should now be editing that label, instead of creating a new one.

closed 2022-06-18 by gregdavill