gregdavill/KiBuzzard #69: Error when editing multiline label

elekgeek, 2022-01-11:
I am trying to edit a multiline label that contains ```\r\n``` but error appears:

[image]

This error causes kicad to stay in the process list and prevents it closing cleanly, so I had to kill it manually.

gregdavill, 2022-01-11:
I was expecting some of the editing features to cause issues like this. Thanks for finding a case which breaks it!

I'm trying to work out the best way to handle this. One option is to encode json parameters into a ascii/hex blob, because it should only be edited by KiBuzzard.

closed 2022-01-12 by gregdavill

elekgeek, 2022-01-19:
Greg, 

your latest changes are causing KiCAD to crash. I am trying to find a way to reproduce this disaster :D in a logical manner :D

elekgeek, 2022-01-19:
Seems there is a problem, look at this please:

```
01-19 21:40:04 root 1331:Loading system fonts.
01-19 21:40:04 root 1331:Loading system fonts.
01-19 21:40:04 root 1349:  Found 193 fonts in system
01-19 21:40:04 root 1349:  Found 193 fonts in system
01-19 21:40:11 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:11 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:11 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:11 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:11 root 1251:Unsuported character in <text> element "
"
01-19 21:40:11 root 1251:Unsuported character in <text> element "
"
01-19 21:40:11 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:11 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:11 root 458:  Inlining 10 segments...
01-19 21:40:11 root 458:  Inlining 10 segments...
01-19 21:40:11 root 928:  Writing Text with 922 points
01-19 21:40:11 root 928:  Writing Text with 922 points
01-19 21:40:11 root 928:  Writing Text with 45 points
01-19 21:40:11 root 928:  Writing Text with 45 points
01-19 21:40:16 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:16 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:16 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:16 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:16 root 1251:Unsuported character in <text> element "
"
01-19 21:40:16 root 1251:Unsuported character in <text> element "
"
01-19 21:40:16 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:16 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:16 root 458:  Inlining 1 segments...
01-19 21:40:16 root 458:  Inlining 1 segments...
01-19 21:40:16 root 928:  Writing Text with 132 points
01-19 21:40:16 root 928:  Writing Text with 132 points
01-19 21:40:16 root 928:  Writing Text with 129 points
01-19 21:40:16 root 928:  Writing Text with 129 points
01-19 21:40:16 root 928:  Writing Text with 129 points
01-19 21:40:16 root 928:  Writing Text with 129 points
01-19 21:40:16 root 928:  Writing Text with 122 points
01-19 21:40:16 root 928:  Writing Text with 122 points
01-19 21:40:16 root 928:  Writing Text with 122 points
01-19 21:40:16 root 928:  Writing Text with 122 points
01-19 21:40:16 root 928:  Writing Text with 103 points
01-19 21:40:16 root 928:  Writing Text with 103 points
01-19 21:40:16 root 928:  Writing Text with 86 points
01-19 21:40:16 root 928:  Writing Text with 86 points
01-19 21:40:16 root 928:  Writing Text with 55 points
01-19 21:40:16 root 928:  Writing Text with 55 points
01-19 21:40:16 root 928:  Writing Text with 52 points
01-19 21:40:16 root 928:  Writing Text with 52 points
01-19 21:40:16 root 928:  Writing Text with 19 points
01-19 21:40:16 root 928:  Writing Text with 19 points
01-19 21:40:18 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:18 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:18 root 458:  Inlining 1 segments...
01-19 21:40:18 root 458:  Inlining 1 segments...
01-19 21:40:18 root 928:  Writing Text with 132 points
01-19 21:40:18 root 928:  Writing Text with 132 points
01-19 21:40:18 root 928:  Writing Text with 129 points
01-19 21:40:18 root 928:  Writing Text with 129 points
01-19 21:40:18 root 928:  Writing Text with 129 points
01-19 21:40:18 root 928:  Writing Text with 129 points
01-19 21:40:18 root 928:  Writing Text with 122 points
01-19 21:40:18 root 928:  Writing Text with 122 points
01-19 21:40:18 root 928:  Writing Text with 122 points
01-19 21:40:18 root 928:  Writing Text with 122 points
01-19 21:40:18 root 928:  Writing Text with 103 points
[... 113 lines cut ...]
01-19 21:40:42 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:42 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:42 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:42 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-19 21:40:42 root 1251:Unsuported character in <text> element "
"
01-19 21:40:42 root 1251:Unsuported character in <text> element "
"
01-19 21:40:42 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:42 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:42 root 458:  Inlining 10 segments...
01-19 21:40:42 root 458:  Inlining 10 segments...
01-19 21:40:42 root 928:  Writing Text with 922 points
01-19 21:40:42 root 928:  Writing Text with 922 points
01-19 21:40:42 root 928:  Writing Text with 45 points
01-19 21:40:42 root 928:  Writing Text with 45 points
01-19 21:40:43 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:43 unfiltered 871:Found SVG layer: F.SilkS
01-19 21:40:43 root 458:  Inlining 10 segments...
01-19 21:40:43 root 458:  Inlining 10 segments...
01-19 21:40:43 root 928:  Writing Text with 922 points
01-19 21:40:43 root 928:  Writing Text with 922 points
01-19 21:40:43 root 928:  Writing Text with 45 points
01-19 21:40:43 root 928:  Writing Text with 45 points
```

gregdavill, 2022-01-19:
I wonder if there is a length limit on the footprint attributes, where we're storing the last parameters.

Can you generate a footprint that causes a crash and attach it here? It should be in your clipboard, so just paste it here.

elekgeek, 2022-01-19:
I also noticed that editing the label fw times causes KiCAD to crash as well. Also if a couple of labels centers happen to be at the same coordinates, KiCAD also crashes when I try to either move one of them or even edit one of them as well.

footprint
```
(footprint kibuzzard-61E8921C (layer F.Cu) (tedit 61E8921C) (generator kibuzzard)
    (attr virtual)
    (descr "Converted using: scripting")
    (tags "kb_params=7b27416c69676e6d656e7443686f696365273a20274c656674272c20274361704c65667443686f696365273a20275b272c2027436170526967687443686f696365273a20275d272c2027466f6e74436f6d626f426f78273a20274b474841505059536f6c6964272c20274865696768744374726c273a20272e37272c20274d756c74694c696e6554657874273a2027534f49435c725c6e312e32376d6d272c202750616464696e67426f74746f6d4374726c273a2027322e35272c202750616464696e674c6566744374726c273a2027322e35272c202750616464696e6752696768744374726c273a2027322e35272c202750616464696e67546f704374726c273a2027322e35272c202757696474684374726c273a2027277d")
      (fp_text reference kibuzzard-61E8921C (at 0 -1.2822142288829999) (layer F.SilkS) hide
    (effects (font (size 0.00025400000000000005 0.00025400000000000005) (thickness 2.5400000000000002e-06)))
  )
  (fp_text value G*** (at 0 1.2822142288829999) (layer F.SilkS) hide
    (effects (font (size 0.00025400000000000005 0.00025400000000000005) (thickness 2.5400000000000002e-06)))
  )
  (fp_poly
    (pts 
      (xy -2.308200389286 -1.281706228883)
      (xy -2.308200389286 -0.442119233441)
      (xy -2.307115184208 -0.445374848675)
      (xy -2.298433543583 -0.4540564893)
      (xy -2.288666697879 -0.463823335003)
      (xy -2.277272044559 -0.473047578167)
      (xy -2.25936616077 -0.487155244183)
      (xy -2.240917674442 -0.501262910199)
      (xy -2.227895213504 -0.510487153363)
      (xy -2.220841380497 -0.515370576214)
      (xy -2.208361522098 -0.5235096143)
      (xy -2.195339061161 -0.532191254925)
      (xy -2.185572215458 -0.538702485394)
      (xy -2.181231395145 -0.541415498089)
      (xy -2.175805369754 -0.543043305706)
      (xy -2.174720164676 -0.543043305706)
      (xy -2.17254975452 -0.543043305706)
      (xy -2.11449128284 -0.483899628949)
      (xy -2.056432811161 -0.426926362347)
      (xy -2.02767487659 -0.409563081097)
      (xy -1.994576121708 -0.404137055706)
      (xy -1.971786815067 -0.405764863324)
      (xy -1.954423533817 -0.417159516644)
      (xy -1.954423533817 -0.438863618206)
      (xy -1.975042430301 -0.519168793988)
      (xy -2.027132274051 -0.584281098675)
      (xy -2.042867747684 -0.598388764691)
      (xy -2.069455272098 -0.622805878949)
      (xy -2.102011424442 -0.652106416058)
      (xy -2.134567576786 -0.681949555706)
      (xy -2.1611551012 -0.705824067425)
      (xy -2.175805369754 -0.718846528363)
      (xy -2.219756175418 -0.766052949261)
      (xy -2.257195750614 -0.819770600628)
      (xy -2.283240672489 -0.878914277386)
      (xy -2.293007518192 -0.942398774456)
      (xy -2.27672944202 -1.011851899456)
      (xy -2.237662059208 -1.073708588909)
      (xy -2.2306082262 -1.079677216839)
      (xy -2.223554393192 -1.086731049847)
      (xy -2.182045298954 -1.118337647752)
      (xy -2.13782319202 -1.143704316449)
      (xy -2.089802867313 -1.16038934452)
      (xy -2.036899119754 -1.16595102055)
      (xy -1.959306956668 -1.157269379925)
      (xy -1.895822459598 -1.132309663128)
      (xy -1.838849192997 -1.091614472699)
      (xy -1.781875926395 -1.035726411175)
      (xy -1.763970042606 -1.014564912152)
      (xy -1.758001414676 -0.990147797894)
      (xy -1.758001414676 -0.980380952191)
      (xy -1.766140452762 -0.971699311566)
      (xy -1.777535106083 -0.960847260784)
      (xy -1.795440989872 -0.94619699223)
[... 1142 lines cut ...]
01-20 00:36:10 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-20 00:36:10 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-20 00:36:10 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-20 00:36:10 root 1198:Font "KGHAPPYSolid" does not natively support style "Regular" using "Path" instead
01-20 00:36:10 root 1251:Unsuported character in <text> element "
"
01-20 00:36:10 root 1251:Unsuported character in <text> element "
"
01-20 00:36:11 unfiltered 871:Found SVG layer: F.SilkS
01-20 00:36:11 unfiltered 871:Found SVG layer: F.SilkS
01-20 00:36:11 root 458:  Inlining 10 segments...
01-20 00:36:11 root 458:  Inlining 10 segments...
01-20 00:36:11 root 928:  Writing Text with 922 points
01-20 00:36:11 root 928:  Writing Text with 922 points
01-20 00:36:11 root 928:  Writing Text with 45 points
01-20 00:36:11 root 928:  Writing Text with 45 points
01-20 00:36:13 unfiltered 871:Found SVG layer: F.SilkS
01-20 00:36:13 unfiltered 871:Found SVG layer: F.SilkS
01-20 00:36:13 root 458:  Inlining 10 segments...
01-20 00:36:13 root 458:  Inlining 10 segments...
01-20 00:36:13 root 928:  Writing Text with 922 points
01-20 00:36:13 root 928:  Writing Text with 922 points
01-20 00:36:13 root 928:  Writing Text with 45 points
01-20 00:36:13 root 928:  Writing Text with 45 points
```

elekgeek, 2022-01-27:
Hi Greg,

Any chance on this issue.

thanks

gregdavill, 2022-02-10:
There does appear to be some quirks with the editing. Appears to be with how we're adding/removing footprints.

Things appear to work, but if you attempt to delete an edited footprint, then the graphics for it aren't correctly removed, and attempting to select the ghost footprint then has caused a crash for me.

elekgeek, 2022-02-11:
Exactly!

gregdavill, 2022-02-12:
@elekgeek Because this is a different issue, I've opened up a new issue here: #74 

I've also committed a fix. Can you try it out and see if you hit any issues?

elekgeek, 2022-02-12:
Hi,

Thank you for your time, indeed!

Look at this video, FSilk.S has an issue that you will spot on the fly, and at the end, closing PCB Editor crashes and all KiCAD main window closes, thu main window should not close.. So, the ghost issue is resolved, but crashing is not, sorry man. I am really glad we have layers options now, really thanks for all contributors!

https://user-images.githubusercontent.com/8074127/153710375-64b39ec0-bdb7-45d1-85f6-2c95c119eddc.mp4

commit edf7d267 (2022-01-12): Encode parameters as hex string in footprints

Fixes #69