gregdavill/KiBuzzard #71: Recognized selected text

hildogjr, 2022-01-13:
If one text object (and only) is already selected when activate the plugin, it will be nice to recognize the text, size and other parameters in the dialog GUI.

gregdavill, 2022-01-13:
This is currently implemented with some recent commits to `main`. I'm holding off an making it an official release until it's be a bit more thoroughly tested.

There is a clip here: Is this what you have in mind?
https://github.com/gregdavill/KiBuzzard/pull/65

hildogjr, 2022-02-19:
Yes. Closing this as already implemented.
Waiting the official KiCad package.

gregdavill, 2022-02-19:
Yep, should be pretty complete now. 

If you want to try it out, you can with the git release. I'll be packaging up a new version when this issue is fixed: https://gitlab.com/kicad/code/kicad/-/issues/10827