gregdavill/KiBuzzard #72: Svg2Mod Update and python 2 support

Sodium-Hydrogen, 2022-02-06:
Svg2mod recently released a new version with some performance improvements. svg2mod/svg2mod#43

From my understanding Kicad v6 only uses python 3 now.

Is KiBuzzard committed to continue to release versions for both kicad 5.1 and 6?

If so I will update the python2 branch of svg2mod to be up to date with the latest changes.

-----
Just an FYI the main file in svg2mod was getting very large so I split it into 4 new files.
* cli.py `Argparser and main function`
* svg2mod.py `LineSegment and PolygonSegment classes`
* exporter.py `Svg2ModExport and its children`
* importer.py `Svg2ModImport`

gregdavill, 2022-02-06:
Moving forward now that v6 is out, I think we can make the switch to drop 5.1 support at some point.

I've got some features that I think can be implemented before updating svg2mod so that these features could be available to 5.1 users.

Thanks for letting me know about the update. I think it'll be good to update! Since KiBuzzard makes pretty heavy use of svg2mod for rendering.

Sodium-Hydrogen, 2022-02-07:
Since you plan on adding a couple of features for 5.1 I've gone ahead and updated the [`python2`](https://github.com/svg2mod/svg2mod/tree/python2) branch to include the latest changes so 5.1 users can have the performance improvements.

gregdavill, 2022-02-07:
FYI, the way svg2mod is currently packaged with KiBuzzard, as a subtree, I've not updated it in a along while!

gregdavill, 2022-02-24:
I've pulled in the newest svg2mod code and in https://github.com/gregdavill/KiBuzzard/commit/980be7839a3bdb56437a147e9bbae75f5b5edae4 I've updated KiBuzzard with code changes needed for this.

I've also dropped KiCad v5 support. Mostly due to it making use of python 2. 
(Users of v5 should use the 1.0.0 or 1.1.0 releases of KiBuzzard.)

closed 2022-02-24 by gregdavill