gregdavill/KiBuzzard #78: Letter P doesn't show up in some labels

voltlog, 2022-03-09:
Using Kicad 6.0.1 Kibuzzard 1.0.0 stable, not sure if this is even a Kibuzzard bug but the letter P doesn't show up in board view, while in this arrangement "2P" or "3P". The 3D viewer does show all labels correctly
[image]
[image]

gregdavill, 2022-03-10:
This is Interesting. If you open up the good and bad labels in the footprint editor, do they look different? 

svg2mod is a tool embedded within KiBuzzard which will attempt to insert a cut through overlapping polygons like the letter P here.

voltlog, 2022-03-10:
This is how a "bad" label looks like in the footprint editor:
[image]

voltlog, 2022-03-10:
I've also now tried 1.1.0 and it looks like the problem is still present. Is this something that you are able to replicate on your end?

gregdavill, 2022-03-10:
I'm using KiCad 6.0.2, on Ubuntu. 

With 1.1.0 from the PCM I've not been able to replicate your issue. I can see my labels are generating a very similar looking polygon, here it's just not filled.

[image]
[image]

I was mistaken about the svg2mod updates, they're not included in the 1.1.0 release. The updates to svg2mod place the "cuts" vertical, which might help, if KiCad is for some reason detecting an overlapping polygon, thus not rendering the inner cutouts.

[image]

To rule this out can you try out the current git version? I've created a PCM package here: 
 - Uninstall your current KiBuzzard in the PCM
 - Use "install from file", to install this version: [KiBuzzard-1.2.0-pcm.zip](https://github.com/gregdavill/KiBuzzard/files/8221257/KiBuzzard-1.2.0-pcm.zip)

voltlog, 2022-03-11:
Using Kicad 6.0.1 on Windows 11 here and with the PCM package you created the problem is no longer present.
[image]

closed 2022-06-18 by gregdavill