gregdavill/KiBuzzard #81: Store label settings in generated footprint

robin7331, 2022-04-29:
As long as there is no option to edit label it might be helpful to store the used values (size, font, ...) in some property in the footprint.

If a label has to be edited (e.g. the old one gets deleted and a new one created) one can then take those values to reproduce the look of the original one.

[image]

gregdavill, 2022-04-30:
We probably could. Do you know if these values are available via the python API?

We use python API to read the current parameters when editing. I'd wanted to use these attributes, but couldn't work out how to read/write these.

gregdavill, 2022-04-30:
Also these value can be edited outside of the plugin, so what could happen if a user charges values so they're invalid then tried to edit?

robin7331, 2022-05-01:
I don't know if there's an API for those values. There may be a way to hook into some memory or file based storage KiCad has for its own stuff? I mean once you place a regular component you can edit it afterwards as well. 

Unlikely but what if there was some hook to tell KiBuzzard that properties have been changed by a user so KiBuzzard can redraw?

gregdavill, 2022-05-02:
Ohh I just re-read your first message. Have you used the new "latest" version on the PCM. We have an editing option built in now.

In the PCB try out version 1.1.0, it's marked as a "testing" release because the editing resulting KiCad crashing on close due to undo corruption. (Fixed in 6.4)

gregdavill, 2022-05-02:
See demo video in the PR: https://github.com/gregdavill/KiBuzzard/pull/65

robin7331, 2022-05-19:
Oh, I haven't seen that this is already implemented. 
Great work! 🎉