gregdavill/KiBuzzard #82: 6.04 - Windows 11 - No label is generated.

MariusHeier, 2022-05-01:
Its me from Twitter. OS info in the topic.

MariusHeier, 2022-05-01:
This is my CTRL V output, but nothing happends.

(module kibuzzard-626E85A5 (layer F.Cu) (tedit 626E85A5)
  (attr board_only exclude_from_pos_files exclude_from_bom)
  (descr "Converted using: scripting")
  (tags svg2mod)
  (fp_text reference kibuzzard-626E85A5 (at 0 -2.051425708333) (layer F.SilkS) hide
    (effects (font (size 0.00025400000000000005 0.00025400000000000005) (thickness 2.5400000000000002e-06)))
  )
  (fp_text value G*** (at 0 2.051425708333) (layer F.SilkS) hide
    (effects (font (size 0.00025400000000000005 0.00025400000000000005) (thickness 2.5400000000000002e-06)))
  )
  (fp_poly
    (pts 
      (xy -8.208168429285 -2.050917708347)
      (xy -7.898605929285 -1.4716125)
      (xy -6.322218429285 -1.4716125)
      (xy -5.769768429285 1.4430375)
      (xy -5.769768429285 -1.4335125)
      (xy -5.393530929285 -1.4906625)
      (xy -5.064918429285 -1.50495)
      (xy -4.786907491785 -1.48351875)
      (xy -4.533899679285 -1.419225)
      (xy -4.311848116785 -1.3090921875)
      (xy -4.126705929285 -1.15014375)
      (xy -3.978473116785 -0.941784375)
      (xy -3.867149679285 -0.68341875)
      (xy -3.797498116785 -0.3708796875)
      (xy -3.774280929285 -0.0)
      (xy -3.799879366785 0.3792140625)
      (xy -3.876674679285 0.69770625)
      (xy -3.998713741785 0.9590484375)
      (xy -4.160043429285 1.1668125)
      (xy -3.455193429285 1.31445)
      (xy -3.283743429285 0.8382)
      (xy -2.976562179285 0.981075)
      (xy -2.778323116785 1.03108125)
      (xy -2.540793429285 1.04775)
      (xy -2.295524679285 1.01679375)
      (xy -2.140743429285 0.93106875)
      (xy -2.062162179285 0.80724375)
      (xy -2.040730929285 0.66675)
      (xy -2.093118429285 0.504825)
      (xy -2.226468429285 0.37861875)
      (xy -2.407443429285 0.27860625)
      (xy -2.607468429285 0.1952625)
      (xy -2.883693429285 0.085725)
      (xy -3.143249679285 -0.0714375)
      (xy -3.336130929285 -0.30718125)
      (xy -3.412330929285 -0.657225)
      (xy -3.345060616785 -1.02631875)
      (xy -3.143249679285 -1.304925)
      (xy -2.941372595939 -1.434570833347)
      (xy -2.694516345939 -1.512358333347)
      (xy -2.402680929285 -1.5382875)
      (xy -2.148482491785 -1.522809375)
      (xy -1.928812179285 -1.476375)
      (xy -1.588293429285 -1.3382625)
      (xy -1.759743429285 -0.885825)
      (xy -2.024062179285 -1.000125)
      (xy -2.359818429285 -1.04775)
      (xy -2.619110095939 -1.010179166653)
      (xy -2.774685095939 -0.897466666653)
      (xy -2.826543429285 -0.7096125)
      (xy -2.778918429285 -0.55959375)
      (xy -2.657474679285 -0.44529375)
      (xy -2.490787179285 -0.3571875)
      (xy -2.307430929285 -0.28575)
      (xy -2.019299679285 -0.16906875)
      (xy -1.743074679285 0.00238125)
[... 257 lines cut ...]
    (layer F.SilkS)
    (width 0.0)
  )
  (fp_poly
    (pts 
      (xy -5.183980929285 1.000125)
      (xy -5.131593429285 1.0048875)
      (xy -5.079205929285 1.0048875)
      (xy -4.743449679285 0.9286875)
      (xy -4.524374679285 0.7191375)
      (xy -4.405312179285 0.40243125)
      (xy -4.378523116785 0.2101453125)
      (xy -4.369593429285 -0.0)
      (xy -4.398168429285 -0.371475)
      (xy -4.498180929285 -0.6905625)
      (xy -4.693443429285 -0.91201875)
      (xy -5.012530929285 -0.9953625)
      (xy -5.098255929285 -0.99298125)
      (xy -5.183980929285 -0.981075)
      (xy -5.183980929285 1.000125)
    )
    (layer F.SilkS)
    (width 0.0)
  )
)

gregdavill, 2022-05-02:
Did you install via the KiCad Plugin & content manager?

Can you try installing version 1.1.0? 

[image]

MariusHeier, 2022-05-02:
1.1.0 works fine: ) Thanks

gregdavill, 2022-05-02:
Great! :)

closed 2022-05-02 by gregdavill