gregdavill/KiBuzzard #83: Some ideas for QoL

immortal-sniper1, 2022-05-01:
So I am making a lot of labels so i searched a bit and got the idea these would be good ideas:
-add a key shortcut
-add the ability to save some templates
-kicad often increments stuff with the <INSERT> key maybe we can add something like this for labels lets say it increments the numbers in the tag or with some syntax

gregdavill, 2022-05-02:
Key shortcut would be good! As far as I know this is a KiCad Plugin limitation that plugins can't be bound to keyboard shortcuts yet. (https://gitlab.com/kicad/code/kicad/-/issues/3837)

Templates could be interesting, like having a small history list? Not sure how that look from a UI perspective. 
As a workaround, if you use 1.1.0 then you can edit an existing footprint, this will let you copy/paste a footprint then edit it's text. (With the label selected open the KiBuzzard window to edit)

It might be possible to detect a number at the end of a footprint and increment it if say the `insert` key was pressed in the KiBuzzard window. Unfortunately with a plugin it's not possible to hook into the internal KiCad handling of this.

immortal-sniper1, 2022-05-02:
ok
I installed the latest from kicad not sure what version it is tho if it is 1.1.0 how can i edit them? with `E` like normal entities or is there another process?

gregdavill, 2022-05-03:
Unfortunately no, plugins can't loop into that level of integration with KiCad.

With a footprint selected (Needs to have been created with KiBuzzard 1.1.0) Open up KiBuzzard, it should load in that footprint and on pressing OK update that selected label

gregdavill, 2022-05-03:
You can see a demo video here: https://github.com/gregdavill/KiBuzzard/pull/65

closed 2024-05-06 by gregdavill