gregdavill/KiBuzzard #84: Can't detect in Win10 / KiCAD 5.1.4

stanleyseow, 2022-05-10:
I look thru all the closed issues and still can't figured out why mine can't be detected.

I tried both scripting and plugins folders ...

I downloaded the zip, rename and copied it to 
C:\Users\nnn\AppData\Roaming\kicad\scripting

Current location :-
C:\Users\nnn\AppData\Roaming\kicad\scripting\KiBuzzard

 Directory of C:\Users\nnn\AppData\Roaming\kicad\scripting\KiBuzzard

10/05/2022  11:18 AM    <DIR>          .
10/05/2022  11:18 AM    <DIR>          ..
25/02/2022  10:24 AM                65 .gitignore
10/05/2022  11:18 AM    <DIR>          doc
25/02/2022  10:24 AM    <DIR>          icons
10/05/2022  11:18 AM    <DIR>          KiBuzzard
25/02/2022  10:24 AM             1,092 LICENCE
10/05/2022  11:18 AM    <DIR>          pcm
25/02/2022  10:24 AM             3,159 README.md
25/02/2022  10:24 AM           155,959 text_dialog.fbp
25/02/2022  10:24 AM                29 __init__.py

Under Preferences , pcbnew , action plugins, there was only iBOM plugins ...
Restarted all the Kicad programs ....

Any other suggestions ???

gregdavill, 2022-05-10:
This plugin is no longer being maintained to support KiCad 5.1, you can try downloading version 1.0.0 or 1.1.0 from the Releases, and try that instead, but the current git repo has some changes that can't be supported in KiCad 5.

Is there a particular reason you can't use KiCad 6?

stanleyseow, 2022-05-11:
Got it... Thanks.. will d/l ver 1.1.x

I hv not been using KiCAD for a while, was doing a project so I not want to upgrade it first....

closed 2022-05-11 by stanleyseow