gregdavill/KiBuzzard #86: Adding a `:` or `;` with the Ubuntu font (and other external fonts) breaks output

carlosperate, 2022-06-28:
With the built in font it only happens with the Ubuntu font, but i could replicate with a couple of extra fonts manually added

This GIF shows that the preview breaks, and if you try to create the label it doesn't produce anything:

[image]

Plugin version: 1.2.0
Kicad version:
```
Application: KiCad

Version: (6.0.5-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.77.0 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.42.0

Platform: macOS Version 12.1 (Build 21C52), 64 bit, Little endian, wxMac

Build Info:
	Date: May  3 2022 12:26:25
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.78.0
	OCC: 7.6.2
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

gregdavill, 2022-06-29:
This seems like it might be specific to some fonts. Maybe something about the glyph selection.

gregdavill, 2022-11-22:
This is fixed via https://github.com/gregdavill/KiBuzzard/commit/96dfe5cd91980ea02e8e1e38ed4b99d39a7ce54f

Release 1.3.0-rc, contains this fix.
Which can be installed via the KiCad PCM.
Uninstalling the current version, and then selecting "Install From File..." And selecting `KiBuzzard-1.3.0-pcm.zip` from here https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0-rc

closed 2022-11-22 by gregdavill