gregdavill/KiBuzzard #87: Version 1.2.0 doesn't work on Ubuntu 20.04 and KiCAD 6.0.7

yaqwsx, 2022-08-13:
I just upgraded from version 1.0.0 to 1.2.0, and the plugin doesn't load. The problem is in:

```
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/xmrazek7/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/__init__.py", line 57, in <module>
    from .plugin import KiBuzzardPlugin
  File "/home/xmrazek7/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/plugin.py", line 15, in <module>
    from .buzzard.buzzard import Buzzard
  File "/home/xmrazek7/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/buzzard/buzzard.py", line 11, in <module>
    from svg2mod.exporter import Svg2ModExport, Svg2ModExportLatest, DEFAULT_DPI
  File "/home/xmrazek7/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/buzzard/../deps/svg2mod/svg2mod/exporter.py", line 33, in <module>
    from svg2mod.coloredlogger import logger, unfiltered_logger
ImportError: cannot import name 'logger' from 'svg2mod.coloredlogger' (/home/xmrazek7/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/buzzard/../deps/svg2mod/svg2mod/coloredlogger.py)
```

I tried to debug it, but there's something I don't understand as `svg2mod.coloredlogger` clearly defines `logger`.

gregdavill, 2022-08-17:
Curious. I'll take a look through that file too. 
But I've not seen this issue. I am however using a local git copy, rather than from the PCM.

gregdavill, 2024-05-06:
This issue is quite a few versions behind, So I'm closing it. Feel free to open a new issue using the latest version if you still have problems

closed 2024-05-06 by gregdavill