gregdavill/KiBuzzard #88: Misshaped character 'B' with Roboto Font

timrosskamp, 2022-08-16:
The character 'B' with the Font "Roboto" seems to be problematic. This could also be a problem with the font itself, but the Roboto Font is very popular, so it might be worth looking into it.

Roboto Font: https://fonts.google.com/specimen/Roboto

[image]
[image]

yaqwsx, 2022-08-16:
I am experiencing similar problems with UbuntuMono-B font (the T is cut):

[image]

gregdavill, 2022-08-17:
This looks to be a regression.
There were some performance improvements in svg2mod, with respect to how the polygons are sliced. These changes might be the cause of this type of rendering error.

electrified, 2022-09-28:
Hi, Using the font Exo Black available here https://fonts.google.com/specimen/Exo

I see the same behaviour in version 1.0.0, 1.1.0 and 1.2.0.

Expected
[image]

Actual
[image]

Sodium-Hydrogen, 2022-11-07:
This is related to svg2mod/svg2mod#51.
I am aware and have a fix (svg2mod/svg2mod#53).
I am just finalizing polygon support for svgs then that fix will be merged into svg2mod.

Sodium-Hydrogen, 2022-11-08:
Version 1.2 of svg2mod is out with the fix for this.

I did rework the way styles are handled to support svgs without explicitly defined styles which does seem to break kibuzzard but this change in [buzzard.py 156](https://github.com/gregdavill/KiBuzzard/blob/00168be6342af39940274283d46ba4e55a08c224/KiBuzzard/buzzard/buzzard.py#L156) seems to fix the crashing I was experiencing:

```diff
- offset_text.style += "stroke-width:0.2;"
+ offset_text.style["stroke-width"] = 0.2
```

gregdavill, 2022-11-20:
@Sodium-Hydrogen I've updated the svg2mod dependency (https://github.com/gregdavill/KiBuzzard/pull/94). However when I try the example from above I get a different error, it seems the polygon ends up as self-intersecting. Or atleast ends up with overlapping parallel lines which seem to cause some issues.
[image]
[image]

In the PCB editor the second cutout is missing. However in the 3d view it's all okay.

[image]

I'll poke into this a bit more myself, but thought I'd mention it in case you've seen it before and there is an easy fix.

gregdavill, 2022-11-22:
I'll think about how to fix this in a more elegant way. But a quick fix for the `B` is to flip the order the segments are inlined.

```diff
diff --git a/KiBuzzard/deps/svg2mod/svg2mod/svg2mod.py b/KiBuzzard/deps/svg2mod/svg2mod/svg2mod.py
index f6e63dfd9..92c96cce0 100755
--- a/KiBuzzard/deps/svg2mod/svg2mod/svg2mod.py
+++ b/KiBuzzard/deps/svg2mod/svg2mod/svg2mod.py
@@ -303,7 +303,7 @@ class PolygonSegment:
 
         logger.debug( "  Inlining {} segments...".format( len( segments ) ) )
 
-        segments.sort(reverse=True, key=lambda h: h.bbox[1].y)
+        segments.sort(reverse=False, key=lambda h: h.bbox[1].y)
 
         all_segments = segments[ : ] + [ self ]
         insertions = []
```

This achieves a much cleaner inlining, without the knot that's currently created. (manually separated here to highlight the insertion points)
[image]

[image]

I'm not seeing any downsides to the sorting reversal, unless it tried to perform an insertion to the top, rather than the bottom, then we're back where we started.

Sodium-Hydrogen, 2022-11-22:
Nice patch!

I think your fix works for now, but I'm working on svg even-odd support so I will have to get this working for 100% of cases.

Looking at the English alphabet I think your fix will achieve enough functionality that it will be unlikely anyone else will run into this problem before a comprehensive solution exists.

gregdavill, 2022-11-22:
@timrosskamp Can you confirm that this fix works? It's currently included in a 1.3.0-rc release you can install https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0-rc

closed 2024-05-06 by gregdavill